ARMmbed/mbed-os #12107: Problem with enable_output and enable_input functions in Serial API class

pgasiorMobica, 2019-12-16:
### Description of defect

When using enable_input and enable_output functions with the `Serial` API class, an ambiguity error appears during the compilation. I tried to use those functions during my experiments with sleep and deep sleep. This is caused by inheritance from `SerialBase` and `FileHandle` classes. All code is used in baremetal profile.

```
[Error] main.cpp@19,12: request for member 'enable_output' is ambiguous
[ERROR] ..\main.cpp: In function 'int main()':
..\main.cpp:19:12: error: request for member 'enable_output' is ambiguous
   19 |         pc.enable_output(false);
      |            ^~~~~~~~~~~~~
In file included from ../mbed-os/platform/FileLike.h:21,
                 from ../mbed-os/platform/Stream.h:21,
                 from ../mbed-os/drivers/Serial.h:24,
                 from ../mbed-os/mbed.h:65,
                 from ..\main.cpp:6:
../mbed-os/platform/FileHandle.h:254:17: note: candidates are: 'virtual int mbed::FileHandle::enable_output(bool)'
  254 |     virtual int enable_output(bool enabled)
      |                 ^~~~~~~~~~~~~
In file included from ../mbed-os/drivers/Serial.h:25,
                 from ../mbed-os/mbed.h:65,
                 from ..\main.cpp:6:
../mbed-os/drivers/SerialBase.h:180:10: note:                 'void mbed::SerialBase::enable_output(bool)'
  180 |     void enable_output(bool enable = true);
      |          ^~~~~~~~~~~~~

```

I found three ways of fixing this issue:
* Use explicit source of function:
```
pc.SerialBase::enable_output(false);
```
* Add `using` to the Serial.h class declaration:
```
using SerialBase::enable_input;
using SerialBase::enable_output;
```
* Add appropriate virtual functions in `Serial` class like in `UARTSerial`.

In Serial.h:
```
virtual int enable_input(bool enabled);
virtual int enable_output(bool enabled);
```
and in Serial.cpp:
```
int Serial::enable_input(bool enabled)
{
    SerialBase::enable_input(enabled);

    return 0;
}

int Serial::enable_output(bool enabled)
{
    SerialBase::enable_output(enabled);

    return 0;
}
```

Another problem is that the `pc.enable_output(false);` doesn't work on its own. There is still a possibility to send via this serial after this call. We need to call `pc.enable_input(false);` first and then disable output to block sending. Thus, sending is blocked only when the peripheral is disabled. 
```
pc.printf("Disabling serial output ...");
pc.enable_output(false);
pc.printf("This message is visible but it should not be.");
pc.enable_input(false);
pc.printf("This message is NOT visible.");
```

#### Target(s) affected by this defect ?
NUCLEO_L073RZ, NUCLEO_L432KC and probably any other target.

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os SHA: d128ff488235d806bbefb21ebb00aea0f55a4bb7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed-cli - 1.8.3
Mbed studio - 0.7.0

#### How is this defect reproduced ? 
Create a `Serial` object.
```
Serial pc(SERIAL_TX, SERIAL_RX);
```
Try to test proper working of enable_output function. 
```
pc.printf("Disabling serial output ...");
pc.enable_input(false);
pc.enable_output(false);
pc.printf("This message should not be visible.");
```
And now the error is returned by the compiler.

kjbracey-arm, 2019-12-16:
>  Add using to the Serial.h class declaration:
> `using SerialBase::enable_input;`

This is the correct solution. Patch welcome!

> Another problem is that the pc.enable_output(false); doesn't work on its own. There is still a possibility to send via this serial after this call. We need to call pc.enable_input(false); first and then disable output to block sending. Thus, sending is blocked only when the peripheral is disabled.

Maybe documentation needs to be clearer, but actually muting input/output is not the intent of the call, and is not the way it is implemented so far. It's there to be a power-saving optimisation, indicating that we don't need the facility. I would say that behaviour is undefined if input or output is attempted while disabled. Some devices may block with errors, some might work regardless, maybe depending on other factors, some might crash.

I'm not sure it's worth spending code size trying to make it be an actual "mute" feature.

pgasiorMobica, 2019-12-16:
Thank you for comments!

Created PR #12118 with patch.

closed 2019-12-16 by pgasiorMobica

wdx04, 2020-06-23:
This problem is back in Mbed OS 6's UnbufferedSerial class.

0xc0170, 2020-06-23:
@wdx04 Thanks for the report, can you create a new issue with details?

petertorelli, 2020-12-15:
@wdx04 

Two lines need to be added to UnbufferedSerial.h:

~~~
     *
     * @returns             bitmask of poll events that have occurred.
     */
    short poll(short events) const override;

    using SerialBase::enable_input;
    using SerialBase::enable_output;
    using SerialBase::readable;
    using SerialBase::writeable;
    using SerialBase::format;
~~~