ARMmbed/mbed-os #12115: STM32WB55 - bluetooth BLEGattServer, change mac address of device - setAddress()

pawelm87, 2019-12-16:
### Description of defect
How can I change MAC address in STM32WB55 chip (device must be public). When I tried to do this, device has still get random address. 

#### Target(s) affected by this defect ?
STM32WB55

#### Toolchain(s) (name and version) displaying this defect ?
MbedStudio 0.7.0 + ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 5.14.0
Commit: b6e5a0a8afa34dec9dae8963778aebce0c82a54b
Imported from: https://github.com/ARMmbed/mbed-os

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
MbedStudio 0.7.0

#### How is this defect reproduced ? 
import BLE_GattServer example;
in "BLEProcess.h" file, when the device is connected display mac address:
```
void when_connection(const Gap::ConnectionCallbackParams_t *connection_event)
    {
        printf("Connected.\r\n");

        BLE &ble = _ble_interface;
        uint8_t address[6];
        BLEProtocol::AddressType_t typeP;
        ble.gap().getAddress(&typeP, address);
        tr_info("when_connection(); address: %s, type: %d", tr_array(address, 6), typeP);
    }
```
in function 
`bool set_advertising_parameters()`
I added piece of code to change and display new mac address:
```
    bool set_advertising_parameters()
    {
        Gap &gap = _ble_interface.gap();
        ble_error_t error = gap.setAdvertisingParameters(
            ble::LEGACY_ADVERTISING_HANDLE,
            ble::AdvertisingParameters()
        );

        // change mac address
        uint8_t read_address[6];
        BLEProtocol::AddressType_t typeP;
        gap.getAddress(&typeP, read_address);
        tr_info("set_advertising_parameters(); read original mac address: %s, type: %d", tr_array(read_address, 6), typeP);

        BLEProtocol::AddressType_t type = BLEProtocol::AddressType_t::PUBLIC;
        BLEProtocol::AddressBytes_t address[BLEProtocol::ADDR_LEN] = {0x80, 0xE1, 0x26, 0x00, 0x44, 0x8D};

        error = gap.setAddress(type, (uint8_t*)address);
        tr_info("set_advertising_parameters(); check error during change address: %d", error);

        gap.getAddress(&typeP, read_address);
        tr_info("set_advertising_parameters(); read NEW mac address: %s, type: %d", tr_array(read_address, 6), typeP);
        //end

        if (error) {
            printf("Gap::setAdvertisingParameters() failed with error %d", error);
            return false;
        }
        return true;
    }
```
My logs in terminal:
```
startBle process started.
Ble instance initialized
[INFO][BLEP]: set_advertising_parameters(); read original mac address: 5e:5e:6b:48:b8:f6, type: 1
[INFO][BLEP]: set_advertising_parameters(); check error during change address: 0
[INFO][BLEP]: set_advertising_parameters(); read NEW mac address: 8d:44:00:26:e1:80, type: 0
[INFO][BLEP]: set_advertising_data(); error set_advertising_data(): 0
Advertising started.
Adding demo service
clock service registered
service handle: 11
        hour characteristic value handle 13
        minute characteristic value handle 15
        second characteristic value handle 17
Connected.
[INFO][BLEP]: when_connection(); address: 8d:44:00:26:e1:80, type: 0
```
I changed MAC address and I read it - is OK. But in the ble scanner my device has old/original mac address before changing. I dont have any error when change mac address (status 0). And that is the problem.

I know that function setAddress is deprecated, so my question is how change correctly change mac address and device must by public ?

LMESTM, 2019-12-17:
Hi @pawelm87 

From what I read here: 
https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/source/generic/GenericGap.tpp#L480
you cannot set the device public address

In STM32WB55RG the public address should be read from Unique device number or from OTP - this is done here:
https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/targets/TARGET_STM/TARGET_NUCLEO_WB55RG/stm32wb_HCIDriver.cpp#L861

pawelm87, 2019-12-19:
@LMESTM what I see the GattServer [example](https://os.mbed.com/docs/mbed-os/v5.14/apis/gattserver.html) work as RANDOM_STATIC in default. I try change to PUBLIC but I can do in this:
```
BLEProtocol::AddressType_t type = BLEProtocol::AddressType_t::PUBLIC;
BLEProtocol::AddressBytes_t address[BLEProtocol::ADDR_LEN] = {0x80, 0xE1, 0x26, 0x00, 0x44, 0x8D};
error = gap.setAddress(type, (uint8_t*)address);
```

but address dont change form random to taken from device :/ How can I do this ? I want to have mac address from device not random.

LMESTM, 2019-12-19:
Again, by reading the comment in the stack, my understanding is that you cannot define the public address by yourself
@pan-  can you help here ?

0xc0170, 2020-02-24:
> @pan- can you help here ?

@ARMmbed/mbed-os-pan Please review

pan-, 2020-09-17:
@pawelm87  We deprecated the setAddress API for two reasons: 
- **Correctness**: From a Bluetooth standpoint it is not correct to change the public address of a device. For random static addresses, it must remain the same until the next controller reset.   
- **Fragmentation**: Because these operations are not standard not all controller offers the same capabilities and code for one target may not work on another target even if they target the same Bluetooth standard. 

It would be useful for us if you could detail your use case and explain what you are trying to achieve by changing the MAC address. .

pan-, 2020-10-29:
@pawelm87 I'm closing this since we haven't heard back from you. The function `setAddress` has been removed from the Bluetooth API with Mbed OS 6. Changing the public address depends on the target and is not a standard Bluetooth feature. 
Please refer to vendor controller documentation to review how this can be achieved if it's still needed.

closed 2020-10-29 by pan-