ARMmbed/mbed-os #12129: thread.start() call not compiling

surajdagar, 2019-12-17:
Hi,

I am trying to create a new thread as shown in below code. I am getting compilation errors 
Error[Pe304]: no instance of overloaded function "rtos::Thread::start" matches the argument list 

MBED OS - 5.14.1
Tool Chain - IAR 8.4
Target mcu - nRF52840_DK

-------------------Code snapty--------------------
#include "rtos.h"
Thread gps_thread;

void GPS_Manager_Service::gps_Process()
{       
    while(1){}
}  
bool GPS_Manager_Service::init(PinName pn)
{
      gps_thread.start(gps_Process); //Getting Error on this line
      return true;
}

kjbracey-arm, 2019-12-19:
Is `GPS_Manager` a class? If so, I'm not sure how you get that specific error.

It's not sufficient to just pass a member function pointer to a `Callback` - you also need to tell it which actual object to call that method on.

    gps_thread.start({this, &GPS_Manager_Service::gps_Process});

should work.

surajdagar, 2019-12-19:
I am getting below error after modification as you suggest.       

Error[Pe304]: no instance of overloaded function "mbed::callback" matches the argument list C:\mHealth_Apps\SCM_mHealth\mbed-os\rtos\Thread.h 281 
              argument types are: (GPS_Manager_Service *, void (*)()) 
 detected during instantiation of "inline int32_t rtos::Thread::start(T *, M) [with T=GPS_Manager_Service, M=void (*)()]" at line 135 of "C:\mHealth_Apps\SCM_mHealth\gnss\GPS_Manager_Service.cpp"

kjbracey-arm, 2019-12-19:
So is `gps_Process` a static method? The form I gave would be correct for a non-static one. If it's static, I'm not sure what the problem is.

I think I need to see some more of the declaration of the `GPS_Manager_Service`  to understand.

The error messages you're showing don't seem consistent with the code you're showing in either case.

Did you perhaps miss the `{}` braces in my suggestion?

surajdagar, 2019-12-19:
Yes, It was a static method. I make it non-static, It works. Thanks @kjbracey-arm

closed 2019-12-19 by surajdagar

kjbracey-arm, 2019-12-20:
Excellent. I'm still puzzled why your original code didn't work when the method was static though. I think it should have done.

Maybe `&gps_Process` would have worked?