ARMmbed/mbed-os #12130: NRF52840 GPIOTE randomly fails to configure input

thendrickson, 2019-12-17:
### Description of defect

gpio_api.c does not initialize the skip_gpio_setup flag so sometimes the setup is skipped. gpio_api.c ought to always explicitly set skip_gpio_setup to false.

nrfx_gpiote_in_config_t is the struct with the skip_gpio_setup field.
All usages of nrfx_gpiote_in_config_t and nrfx_gpiote_in_init() should be reviewed for correct initialization of skip_gpio_setup. There are several other defines meant to be used with nrfx_gpiote_in_init() that are also missing initialization of this field.

Including:
NRFX_GPIOTE_CONFIG_IN_SENSE_LOTOHI
NRFX_GPIOTE_CONFIG_IN_SENSE_HITOLO
NRFX_GPIOTE_CONFIG_IN_SENSE_TOGGLE

So I see a total of 4 bugs here:
1 in gpio.api.c and 3 in nrfx_gpiote.h
targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_NRF52\gpio_api.c
targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_SDK_15_0\modules\nrfx\drivers\include\nrfx_gpiote.h

#### Target(s) affected by this defect ?

Nordic NRF52840 and likely other Nordic MCU devices

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

b6e5a0a8afa34dec9dae8963778aebce0c82a54b

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.1

#### How is this defect reproduced ? 

Use an InterruptIn with a PullUp. You'll see that sometimes the PullUp does not get configured because skip_gpio_setup gets randomly assigned to true sometimes (based on what is on the stack).

0xc0170, 2020-01-08:
cc @ARMmbed/mbed-os-pan

0xc0170, 2020-01-08:
@thendrickson Can you send a pull request with a fix?

cc @desmond-blue

loverdeg-ep, 2020-01-08:
@ARMmbed/team-embeddedplanet

loverdeg-ep, 2020-01-08:
This could potentially be causing a lot of frustration for people trying to reduce power consumption for these parts...

loverdeg, 2020-11-22:
Is this still valid?

loverdeg, 2020-11-22:
@AGlass0fMilk