ARMmbed/mbed-os #12137: Dependency conflict in Mbed OS 5.11 requirements.txt

korjaa, 2019-12-19:
### Description of defect

Cannot install requirements.txt from Mbed OS 5.11 because dependency conflict.
```
ERROR: mbed-greentea 1.6.5 has requirement mbed-os-tools==0.0.6, but you'll have mbed-os-tools 0.0.12 which is incompatible.
```

#### Target(s) affected by this defect ?
All targets

#### Toolchain(s) (name and version) displaying this defect ?
All toolchains

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.11

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.2 and also tried with 1.9.1

#### How is this defect reproduced ? 
```(venv3) jaakor01@e107447:~/workspace/temp
$ pip install mbed-cli
Collecting mbed-cli
  Using cached https://files.pythonhosted.org/packages/05/9e/eca848d298700aba51d67b2fa4933106f23420436e6ad0e549ed9b78ccae/mbed_cli-1.10.2-py2.py3-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/88/74/d5/bed76535d7c4d8c9ec04e34b038b5f2c5a3d88600699301c42/mercurial-5.2.1-cp36-cp36m-linux_x86_64.whl
Collecting pyserial<4.0,>=3.0
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting mbed-os-tools<0.1.0,>=0.0.9
  Using cached https://files.pythonhosted.org/packages/89/4b/300579bd26825fde7293ff5c0114f1045b653bc84aaa670895764866c0e0/mbed_os_tools-0.0.12-py2.py3-none-any.whl
Collecting intelhex<3.0,>=2.0
  Using cached https://files.pythonhosted.org/packages/bf/77/bf670318b3db325c71e2ac6a90b7bcfdf9fc739b7cf6aebb31715721623e/intelhex-2.2.1-py2.py3-none-any.whl
Collecting fasteners
  Using cached https://files.pythonhosted.org/packages/18/bd/55eb2d6397b9c0e263af9d091ebdb756b15756029b3cededf6461481bc63/fasteners-0.15-py2.py3-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/80/34/1c/3967380d9676d162cb59513bd9dc862d0584e045a162095606/prettytable-0.7.2-cp36-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/c6/c5/65/5b166afb6eac87dc300368ec4d92f39502dd41cdc73056d49e/junit_xml-1.8-cp36-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/8b/99/a0/81daf51dcd359a9377b110a8a886b3895921802d2fc1b2397e/future-0.18.2-cp36-none-any.whl
Collecting beautifulsoup4
  Using cached https://files.pythonhosted.org/packages/3b/c8/a55eb6ea11cd7e5ac4bacdf92bac4693b90d3ba79268be16527555e186f0/beautifulsoup4-4.8.1-py3-none-any.whl
Collecting requests<3.0,>=2.0
  Using cached https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl
Collecting six<2.0,>=1.0
  Using cached https://files.pythonhosted.org/packages/65/26/32b8464df2a97e6dd1b656ed26b2c194606c16fe163c695a992b36c11cdf/six-1.13.0-py2.py3-none-any.whl
Collecting appdirs<2.0,>=1.4
  Using cached https://files.pythonhosted.org/packages/56/eb/810e700ed1349edde4cbdc1b2a21e28cdf115f9faf263f6bbf8447c1abf3/appdirs-1.4.3-py2.py3-none-any.whl
Collecting colorama<0.5,>=0.3
  Using cached https://files.pythonhosted.org/packages/c9/dc/45cdef1b4d119eb96316b3117e6d5708a08029992b2fee2c143c7a0a5cc5/colorama-0.4.3-py2.py3-none-any.whl
Collecting lockfile
  Using cached https://files.pythonhosted.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl
Collecting monotonic>=0.1
  Using cached https://files.pythonhosted.org/packages/ac/aa/063eca6a416f397bd99552c534c6d11d57f58f2e94c14780f3bbf818c4cf/monotonic-1.5-py2.py3-none-any.whl
Collecting soupsieve>=1.2
  Using cached https://files.pythonhosted.org/packages/81/94/03c0f04471fc245d08d0a99f7946ac228ca98da4fa75796c507f61e688c2/soupsieve-1.9.5-py2.py3-none-any.whl
Collecting certifi>=2017.4.17
  Using cached https://files.pythonhosted.org/packages/b9/63/df50cac98ea0d5b006c55a399c3bf1db9da7b5a24de7890bc9cfd5dd9e99/certifi-2019.11.28-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting idna<2.9,>=2.5
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1
  Using cached https://files.pythonhosted.org/packages/b4/40/a9837291310ee1ccc242ceb6ebfd9eb21539649f193a7c8c86ba15b98539/urllib3-1.25.7-py2.py3-none-any.whl
Installing collected packages: mercurial, pyserial, intelhex, six, monotonic, fasteners, PrettyTable, junit-xml, future, soupsieve, beautifulsoup4, certifi, chardet, idna, urllib3, requests, appdirs, colorama, lockfile, mbed-os-tools, mbed-cli
Successfully installed PrettyTable-0.7.2 appdirs-1.4.3 beautifulsoup4-4.8.1 certifi-2019.11.28 chardet-3.0.4 colorama-0.4.3 fasteners-0.15 future-0.18.2 idna-2.8 intelhex-2.2.1 junit-xml-1.8 lockfile-0.12.2 mbed-cli-1.10.2 mbed-os-tools-0.0.12 mercurial-5.2.1 monotonic-1.5 pyserial-3.4 requests-2.22.0 six-1.13.0 soupsieve-1.9.5 urllib3-1.25.7
(venv3) jaakor01@e107447:~/workspace/temp
(venv3) jaakor01@e107447:~/workspace/temp
$ mbed import mbed-os-example-nfc
[mbed] Working path "/home/jaakor01/workspace/temp" (directory)
[mbed] Importing program "mbed-os-example-nfc" from "ssh://git@github.com/ARMmbed/mbed-os-example-nfc" at latest revision in the current branch
[mbed] Adding library "NFC_SmartPoster/mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os" at rev #51d55508e840
[mbed] Adding library "NFC_EEPROM/mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os" at rev #51d55508e840
[mbed] Adding library "NFC_EEPROM/source/target/TARGET_M24SR/eeprom_driver" from "ssh://git@github.com/ARMmbed/mbed-nfc-m24sr" at latest revision in the current branch
[... 74 lines cut ...]
  Using cached https://files.pythonhosted.org/packages/38/f1/0c9b852c5136364c0dd59ee1d89e0d7599bef6c9f1869e0f2e07882253ea/semver-2.9.0-py2.py3-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/55/1b/1a/29f8d1d20b748b3825a0bd52c6a0defdf44aa0d150b5e6df3a/jsonmerge-1.7.0-cp36-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/f2/a6/ea/5b4946dbc3f89159311c25c9515fb9306f90d930561daf87bd/mbed_flasher-0.10.1-cp36-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/68/68/4a/e3f5e0e612f335a3b00ea399b195e9f67150c8d3df5450bc2e/yattag-1.12.2-cp36-none-any.whl
Collecting cffi!=1.11.3,>=1.8
  Using cached https://files.pythonhosted.org/packages/49/72/0d42f94fe94afa8030350c26e9d787219f3f008ec9bf6b86c66532b29236/cffi-1.13.2-cp36-cp36m-manylinux1_x86_64.whl
Requirement already satisfied: setuptools in /home/jaakor01/workspace/temp/mbed-os-example-nfc/venv3/lib/python3.6/site-packages (from protobuf<3.6.0,>=3.5.0->manifest-tool==1.4.6->-r requirements.txt (line 22)) (42.0.2)
Collecting lockfile
  Using cached https://files.pythonhosted.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl
Collecting appdirs<2.0,>=1.4
  Using cached https://files.pythonhosted.org/packages/56/eb/810e700ed1349edde4cbdc1b2a21e28cdf115f9faf263f6bbf8447c1abf3/appdirs-1.4.3-py2.py3-none-any.whl
Collecting fasteners
  Using cached https://files.pythonhosted.org/packages/18/bd/55eb2d6397b9c0e263af9d091ebdb756b15756029b3cededf6461481bc63/fasteners-0.15-py2.py3-none-any.whl
Processing /home/jaakor01/.cache/pip/wheels/f2/9a/90/de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511/pycparser-2.19-py2.py3-none-any.whl
Collecting monotonic>=0.1
  Using cached https://files.pythonhosted.org/packages/ac/aa/063eca6a416f397bd99552c534c6d11d57f58f2e94c14780f3bbf818c4cf/monotonic-1.5-py2.py3-none-any.whl
Building wheels for collected packages: manifest-tool
  Building wheel for manifest-tool (setup.py) ... done
  Created wheel for manifest-tool: filename=manifest_tool-1.4.6-cp36-none-any.whl size=71960 sha256=79bc870cd09ce67e0af9b262ccd90a1668c355241e881338bdee3e01a55bad42
  Stored in directory: /tmp/pip-ephem-wheel-cache-14tr25sh/wheels/d3/1b/05/2198482f7359e95e4d82f8d896246e8f3c3894ae7863275cd8
Successfully built manifest-tool
ERROR: mbed-greentea 1.6.5 has requirement mbed-os-tools==0.0.6, but you'll have mbed-os-tools 0.0.12 which is incompatible.
Installing collected packages: colorama, certifi, urllib3, prettytable, six, junit-xml, pyyaml, jsonschema, future, python-dotenv, python-dateutil, mbed-cloud-sdk, chardet, idna, requests, pyserial, MarkupSafe, Jinja2, intelhex, sortedcontainers, intervaltree, beautifulsoup4, lockfile, appdirs, monotonic, fasteners, mbed-os-tools, mbed-ls, mbed-host-te [... cut ...] .1 intervaltree-2.1.0 jsonmerge-1.7.0 jsonschema-2.6.0 junit-xml-1.8 lockfile-0.12.2 manifest-tool-1.4.6 mbed-cloud-sdk-2.0.1 mbed-flasher-0.10.1 mbed-greentea-1.6.5 mbed-host-tests-1.5.10 mbed-ls-1.7.10 mbed-os-tools-0.0.12 monotonic-1.5 prettytable-0.7.2 protobuf-3.5.2.post1 pyasn1-0.2.3 pycparser-2.19 pyelftools-0.25 pyparsing-2.4.5 pyserial-3.4 python-dateutil-2.8.1 python-dotenv-0.10.3 pyyaml-4.2b1 requests-2.20.1 semver-2.9.0 six-1.11.0 sortedcontainers-2.1.0 urllib3-1.23 yattag-1.12.2
```

madchutney, 2020-01-06:
Hi @korjaa, thank you for raising this issue. We shall add it to our known issues list, but unfortunately we are unable to fix it as the requirements are defined in a released version of Mbed OS for which there are no plans to patch.

However, please try creating a new Python environment and installing the following packages before invoking `pip install -r requirements.txt`.

```
pip install mbed-cli==1.10.0
pip install mbed-host-tests==1.5.5
pip install mbed-ls==1.7.5
```

Please let me know if this work around does not work for you.

closed 2020-01-06 by ciarmcom

korjaa, 2020-01-07:
Hi @madchutney, using the workaround solves the issue for me.

```
pip install mbed-cli==1.10.0
pip install mbed-host-tests==1.5.5
pip install mbed-ls==1.7.5
pip install -r mbed-os/requirements.txt
```