ARMmbed/mbed-os #12149: STM32WB Index file missing

felipeLeast, 2019-12-20:
Hi everyone,

I was trying to port the mbed-bootloader project to my board when I stumble upon a problem regardin the board missing flash configuration on the index.json file. After a lot of research and trying I managed to create a configuration that worked for my flash. But as there's no documentation for this file I don't know for a fact that what I did is corret.

I'll be pasting at the end of this issue the lines added in the index.json and on the target.json files. Can anyone, preferebly from ST, confirm that is correct? I can commit it for the next version if it is, but if it's not can you include for next version and help me fix it?

#### Target(s) affected by this defect ?
NUCLEO_WB5RG

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed cli 1.10.0

#### How is this defect reproduced ? 

Change **index.json** file to include:
"STM32WB55RG": {
        "algorithms": [
            {
                "default": true,
                "file_name": "CMSIS/Flash/STM32WBxx_1024.FLM", 
                "ram_size": null, 
                "ram_start": null, 
                "size": 1048576, 
                "start": 134217728
            }
        ], 
        "family": "STM32WB Series",
        "from_pack": {
            "pack": "STM32WBxx_DFP", 
            "url": "http://www.keil.com/pack", 
            "vendor": "Keil", 
            "version": "1.0.0"
        }, 
        "memories": {
            "IRAM1": {
                "access": {
                    "execute": false, 
                    "non_secure": false, 
                    "non_secure_callable": false, 
                    "peripheral": false, 
                    "read": true, 
                    "secure": false, 
                    "write": true
                }, 
                "default": true, 
                "size": 196608, 
                "start": 536870912, 
                "startup": false
            }, 
            "IRAM2": {
                "access": {
                    "execute": false, 
                    "non_secure": false, 
                    "non_secure_callable": false, 
                    "peripheral": false, 
                    "read": true, 
                    "secure": false, 
                    "write": true
                }, 
                "default": false, 
                "size": 65536, 
                "start": 268435456, 
                "startup": false
            }, 
            "IROM1": {
                "access": {
                    "execute": true, 
                    "non_secure": false, 
                    "non_secure_callable": false, 
                    "peripheral": false, 
                    "read": true, 
                    "secure": false, 
                    "write": false
                }, 
                "default": true, 
                "size": 1048576, 
                "start": 134217728, 
                "startup": true
            }
        }, 
        "name": "STM32WB55RG", 
        "processor": {
            "Symmetric": {
                "core": "CortexM4", 
                "fpu": "SinglePrecision", 
                "mpu": "Present", 
                "units": 1
            }
        }, 
        "sectors": [
            [
                134217728, 
                4096
            ], 
            [
                536833520, 
                528
            ], 
            [
                536838144, 
                8
            ]
        ], 
        "sub_family": "STM32WB55",
        "vendor": "STMicroelectronics:13"
    }, 

Change **targets.json** file on the NUCLEO_WB55RG target the following lines:

"components_add": ["FLASHIAP"],
"device_name": "STM32WB55RG",
"bootloader_supported": true

Thank you,

LMESTM, 2019-12-20:
@JanneKiiskila @jeromecoutant 
hey can you help providing the index.json update for STM32WB family ?

@felipeLeast you should not need to post the change, this should be done regularly, unfortunately for @JanneKiiskila and us this is not a fully automated process ...

jeromecoutant, 2019-12-20:
@MarceloSalazar

felipeLeast, 2019-12-20:
Hi @LMESTM, thank you for your fast reply. Okay, I'll wait for the PR for this target and them will integrate with mbed-bootloader project.

JanneKiiskila, 2019-12-20:
PR available, can you give it a try once you have time?
https://github.com/ARMmbed/mbed-os/pull/12162

felipeLeast, 2019-12-22:
Hi @JanneKiiskila, I can only test for the device STM32WB55RG but a few things that I could notice from the start. Is missing the sectors tag, which gives this error when compiling:

```
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/felipe/mbed/mbed-bootloader" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/felipe/mbed/mbed-bootloader/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
[Warning] @,: Compiler version mismatch: Have 8.2.1; expected version >= 9.0.0 and < 10.0.0
Building project mbed-bootloader (NUCLEO_WB55RG, GCC_ARM)
Scan: mbed-bootloader
[ERROR] 'sectors'
[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 1
       Path: "/home/felipe/mbed/mbed-bootloader"
       Command: "/usr/bin/python3 -u /home/felipe/mbed/mbed-bootloader/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_WB55RG --source . --build ./BUILD/NUCLEO_WB55RG/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

Another thing I was looking at the configuration for the STM32WB and you put on ROM a size of 16Mbytes, when the device has only 1MByte.
An on the RAM section you placed the two RAM sectors in one tag only. (I saw the two sectors on the reference file RM0434 page66).

Other than that is very similar to the one I did on the description of this issue

JanneKiiskila, 2019-12-22:
This file is not generated manually, it's actually generated by a tool - so I can't (and should not) take any credit for the file itself. It is generated by a tool called [arm-cmsis-pack-manager](https://github.com/ARMmbed/cmsis-pack-manager), which does it. It uses the CMSIS-packs published by chip/device vendors from the KEIL system as it's input. I think there is now some issue with it and the sector handling - I think the CI automation is catching that with the STM32L-series, but for some reason that's not spotted here by the CI.

@theotherjimmy - any ideas?

JanneKiiskila, 2019-12-22:
I think this is now having the same root-cause as https://github.com/ARMmbed/mbed-os/issues/11798
We have to either fix this issue or fix the automatically produced `index.json` manually.

closed 2020-01-10 by ciarmcom

0xc0170, 2020-01-10:
I believe this was fixed via #12162

jeromecoutant, 2020-01-10:
I don't think this issue can be closed.
A patch in targets.json file adding:
     "device_name": "STM32WB55RGVx"
is missing

jeromecoutant, 2020-01-13:
> A patch in targets.json file adding: "device_name": "STM32WB55RGVx" is missing

@felipeLeast 
Do you think you could provide the fix ?
Thx

JanneKiiskila, 2020-01-13:
PR done; https://github.com/ARMmbed/mbed-os/pull/12245