ARMmbed/mbed-os #12159: [Nordic] TWI/I2C interrupts never get installed

AGlass0fMilk, 2019-12-20:
I'm looking at the I2C API implementation for the nRF5x targets and I noticed the interrupt handler declarations for the nrfx_twi.c are preprocessored out:

https://github.com/ARMmbed/mbed-os/blob/789a193696cf8f06063d136a3b8d9ffc14509c5b/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/drivers/src/nrfx_twim.c#L650-L664

This is to avoid conflicts with the handlers declared in the nrfx_spi.c that share the same symbol name:
https://github.com/ARMmbed/mbed-os/blob/789a193696cf8f06063d136a3b8d9ffc14509c5b/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/drivers/src/nrfx_spi.c#L420-L439

This would be okay if the platform resource sharing (PRS) system was configured correctly; they wouldn't conflict then, because the IRQ handler for the shared NVIC slot would be set to the PRS handler which would subsequently delegate to the driver that owns the slot at that time.

The PRS system is "enabled", but none of the "boxes" are. The sdk_config in master does not even have the declarations for it:

https://github.com/ARMmbed/mbed-os/blob/789a193696cf8f06063d136a3b8d9ffc14509c5b/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/config/sdk_config.h#L4043-L4045

In contrast with the current nrfx config:
https://github.com/NordicSemiconductor/nrfx/blob/7a4c9d946cf1801771fc180acdbf7b878f270093/templates/nRF52840/nrfx_config.h#L705-L743

Maybe there's some magic going on in the Mbed I2C/SPI API implementations that I'm missing, but I don't see where the proper interrupt is installed to handle TWI/I2C interrupts properly.

I'm actually surprised the I2C port even works. Maybe it just affects asynchronous operation?

I'm working on a patch. If anyone could point me to the "magic", if it exists, that'd be great.

@marcuschangarm @desmond-blue

AGlass0fMilk, 2019-12-20:
@maclobdell

0xc0170, 2020-01-08:
> AGlass0fMilk linked a pull request that will close this issue 19 days ago

👍  hopefully this one will be closed soon

AGlass0fMilk, 2020-01-09:
So I realized I mentioned `TWIM` in the original post but the I2C API implementation actually uses the TWI peripheral. However, it's still the same issue there.

When enabled, the Nordic drivers attempt to acquire the resources (memory map/interrupt vector slot) that are shared with other peripherals (eg: TWI0 and SPI0) using the PRS system. If the resources are available, the PRS system will install the given interrupt handler to the shared vector "box".

Nothing should change functionally once these issues are fixed by #12160 , since the peripheral drivers are dynamically initialized/uninitialized (and subsequently release the PRS resources) when appropriate at the Mbed driver level.

PR #12160 (2020-01-15): nRF52: Properly configure nRF SDK for nRF52-series targets

### Summary of changes 

Properly configure Nordic platform resource sharing (PRS) API so it actually delegates the shared interrupt slot for I2C/SPI to the appropriate peripheral.

Before this, the PRS system is enabled but none of the "boxes" are. This caused only the SPI interrupt handler to be properly installed in the NVIC table.

Also updated serial API to be fully migrated to using the UARTE peripheral driver (as opposed to the deprecated UART driver). Disabled the UART and TWIM drivers as these are not used by either `serial_a [... cut]
closed 2020-01-15 by 0xc0170