ARMmbed/mbed-os #12172: STM serial configured in Simplex causes runtime assert

alkaes, 2019-12-29:
### Description of defect
System halts on assert when trying to declare a serial object that one of the pins declared as NC (according to official Mbed OS reference)
For example: simple declaration of: `Serial tx_only_ser(PE_8, NC);`
**Hangs on assert at:** https://github.com/ARMmbed/mbed-os/blob/bacf37894945a5c9b7619b709eeabeeb68d25d9c/targets/TARGET_STM/pinmap.c#L62
**That is called from:** https://github.com/ARMmbed/mbed-os/blob/bacf37894945a5c9b7619b709eeabeeb68d25d9c/targets/TARGET_STM/serial_api.c#L161
#### Target(s) affected by this defect ?
STM
#### Toolchain(s) (name and version) displaying this defect ?
IAR 8.40
#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.0
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli
#### How is this defect reproduced ?
```
main()
{
 Serial tx_only_serial(PE_8, NC); // halts here
 tx_only_serial.putc('A');
}

```

0xc0170, 2020-01-02:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-01-02:
Hi

@mprse @kjbracey-arm 

I think issue came with static pinmap feature.
Before that pinmap_pinout was called for each uart pin, which is OK when pin is "NC".
Now serial init calls pin_function which goes to assert when pin is "NC".

mprse, 2020-01-03:
Looking into this.

mprse, 2020-01-03:
The patch can be found here: PR https://github.com/ARMmbed/mbed-os/pull/12186.

0xc0170, 2020-01-08:
Fixed via 12186

closed 2020-01-08 by 0xc0170