ARMmbed/mbed-os #12184: LWIP::socket_close() can attempt to free a null pbuf structure

nat45928, 2020-01-02:
### Description of defect

When closing a TCP socket with LwIP debugging enabled I can see a debug statement `pbuf_free(p == NULL) was called.` It appears change 3b178a08efcfb62a8b53c8aa6781e8ab7def6963 modified a statement from `netbuf_delete()` to `pbuf_free()`. The `netbuf_delete()` function includes a null check which is not included in `pbuf_free()`.

#### Target(s) affected by this defect ?

NUCLEO_F767ZI

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM 9-2019q4

#### What version of Mbed-os are you using (tag or sha) ?

de798c4f379adbcf282199ba06553c0fa644cc8b

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

n/a

0xc0170, 2020-01-03:
@kjbracey-arm Please review

kjbracey-arm, 2020-01-07:
I agree, it can do this on `socket_close. Looks fine in the other part of the change in `socket_recv`. I'll make a patch.

PR #12196 (2020-01-07): LWIP: don't pbuf_free(NULL) on socket close

### Summary of changes 

`pbuf_free(NULL)` causes an `LWIP_ASSERT`. Make the call conditional.

Fixes #12184

#### Impact of changes 

#### Migration actions required 

### Documentation 

None

----------------------------------------------------------------------------------------------------------------
### Pull request type
commit 988ef099 (2020-01-07): LWIP: don't pbuf_free(NULL) on socket close

`pbuf_free(NULL)` causes an `LWIP_ASSERT`. Make the call conditional.

Fixes #12184
commit 99344963 (2020-04-27): LWIP: don't pbuf_free(NULL) on socket close

`pbuf_free(NULL)` causes an `LWIP_ASSERT`. Make the call conditional.

Fixes #12184
closed 2020-01-07 by 0xc0170