ARMmbed/mbed-os #12190: Is there a way to include the image header in the generated *_update.bin?

Reda-RM, 2020-01-05:
### Description of defect

While updating the firmware over the air, we need to send the image header + the image itself. The build artifacts include *_update.bin which should be used to be sent to the device over the air. However this binary includes only the image without a header. Is there a clean way to include the image header in this binary? Currently I am editing ./tools/regions.by as follows to include the image header:
```
--- a/tools/regions.py
+++ b/tools/regions.py
@@ -33,6 +33,7 @@ from .utils import (
 from intelhex import IntelHex
 
 UPDATE_WHITELIST = (
+    "header",
     "application"
 )
```

#### Target(s) affected by this defect ?
Any

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-01-08:
cc @ARMmbed/mbed-os-tools

0xc0170, 2020-02-24:
It used to be there but can't find a reason why it was removed.

I tracked this to https://github.com/ARMmbed/mbed-os/pull/7714/commits/a7ddea78d71dd12630698ae4a302f2da291d2180. But I can't find a reason in the PR itself neither in the commit - why is this being removed (useful to provide a reason to a commit itself but we fail sometimes to do so 😞 ).

cc @ARMmbed/mbed-os-core 

@teetak01 If you happen to know by any chance

teetak01, 2020-02-24:
This is how Pelion update is using the images. Is this related to managed vs. unmanaged bootloader mode?

SeppoTakalo, 2020-02-25:
Appending the header must have been removed, because Pelion does not use it, and they require separate manifest file.
The header is then generated by the update client, once the image is downloaded to the device.
Bootloader then verifies this header to check whether the image is correct.

But, please note that if your linker file and `mbed_app.json` are correctly set, you will get four images. For example, building a `hello` application:
* `hello_application.bin` which is the raw application image after a linking phase. On some devices, this is in `.hex` format.
* `hello_header.bin` which is the update header.
*  `hello_update.bin` which is the "update image" but actually this is identical to application on most devices. This is always in `.bin` format.
*  `hello.bin`, which is the `_header.bin` + `_application.bin`, so exactly what you requested.

0xc0170, 2020-02-26:
Thanks @SeppoTakalo . @Reda-RM Does that answer your question?

0xc0170, 2020-03-02:
I'll close this as resolved

closed 2020-03-02 by 0xc0170

Reda-RM, 2020-03-04:
@SeppoTakalo  in my case, hello.bin contains the bootloader as well (bootloader + header + image). So I can not generate an image which contains only the (the header + image) to use in firmware update. what do you think ?

SeppoTakalo, 2020-03-04:
How about:
```
cp ./BUILD/<target>/<toolchain/hello_header.bin image.bin
dd if=./BUILD/<target>/<toolchain/hello_app.bin of=image.bin bs=1 seek=1024
```

or any other way to append those together...
Header have always space of 1 kB allocated.

Reda-RM, 2020-03-04:
Yeah if we are going to make it manual then this way will work. I was asking if it can be done automatically in build process. 
Thanks @SeppoTakalo

phlegx, 2021-06-18:
I have only a `hello_header.hex`. And now?