ARMmbed/mbed-os #12191: BLE connectivity issues with iPhone 11

sr425, 2020-01-05:
### Description of defect

Using the mbed BLE_LED sample on a Nordic NRF52_DK a iPhone 11 Pro (and other iOS devices like the current generation iPad) one gets an immediate timeout after connecting to the device. The nRF Connect app reports an "unexpected timeout". 
This behaviour seems to be independent of the iOS version, since an older iPad Pro with the same iOS version (13.3) works fine. 
Unfortunately I have currently issues sniffing the BLE traffic itself.

#### Target(s) affected by this defect ?

NRF52_DK

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM 9 2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

5.13.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli: 1.10.1
Toolchain: GCC, version:  9 2019-q4-major (9.2.1 20191025)

#### How is this defect reproduced ? 

Use the mbed-os BLE_LED sample, then compile and flash with 
_mbed compile -m NRF52_DK -t GCC_ARM --flash_
After that try connecting with a current generation iPhone or iPad (e.g. iPhone 11 Pro) and the nRF Connect app. Immediately after connecting a timeout occurs, which is not the case with older devices.

trowbridgec, 2020-01-06:
@AGlass0fMilk

AGlass0fMilk, 2020-01-06:
@sr425 Can you post any debug uart output you see from the Mbed device while trying to connect?

If you have the ability to also do BLE traces on the iPhone (probably requires iOS developer license) that would also be helpful.

sr425, 2020-01-06:
@AGlass0fMilk Currently I am trying to get the wanted data, but having some issues with UART output. Is there anything special regarding the NRF52DK to get the output?
Regarding the iPhone traces, I have some syslog with BLE profile, what files from this log is interesting in particular and how to open it?

AGlass0fMilk, 2020-01-06:
You may need to enable hardware flow control on your serial port to get the debug UART to show up.

I'm not familiar with iPhone debug logs so I can't tell you what would be of interest. Mainly anything that shows how the Mbed BLE device was behaving as seen by the iPhone BLE stack -- that would be useful to better understand where the issue is.

0xc0170, 2020-01-08:
cc @ARMmbed/mbed-os-pan

sr425, 2020-01-12:
@AGlass0fMilk Could you give me some hints on how to do that, since I still have issues getting the UART output?

AGlass0fMilk, 2020-01-13:
@sr425 What terminal program are you using?

sr425, 2020-01-13:
Bash on Ubuntu and ZSH on MacOS

AGlass0fMilk, 2020-01-13:
I mean for viewing the debug UART output. I am most familiar with minicom.

sr425, 2020-01-13:
Ah okay, I can also try to use minicom if this makes helping easier.

AGlass0fMilk, 2020-01-13:
On Ubuntu:
1. Run `ls /dev/tty*` to list tty devices connected to your computer
2. Plug in Mbed board (nRF52DK in this case)
3. Run `ls /dev/tty*` again, note the additional device (probably something like `/dev/ttyACM0`). This will be the serial port you want to access to get debug/printf output from the Mbed board
4. Install minicom: `sudo apt-get install minicom`
5. Start minicom with settings menu open: `minicom -s`
6. Scroll down to "Serial port setup" in the menu and press enter
7. Press `A` and change the "Serial Device" to whatever you found in step 3 then press enter
8. Make sure "Bps/Par/Bits" is set to "9600 8N1", if not press `E` and select that option
9. Press `F` until "Hardware Flow Control" option says "Yes"
10. Press enter. You should see debug output from the Mbed board whenever "printf" is used in the code.

maclobdell, 2020-02-06:
@AGlass0fMilk - any updates on this from your side?

sr425, 2020-02-13:
We invested quite some time, but could not find anything relevant in the (very long) iOS trace and also nothing relevant was shown on the serial output.

desmond-blue, 2020-02-14:
@sr425 Still no OTA trace can be shared?

desmond-blue, 2020-03-03:
@sr425 As you are using `mbed-os-5.13.1`, the default BLE Stack is Nordic SoftDevice, could you confirm is that you are using? Or you've already configured it to using Cordio?

sr425, 2020-03-05:
Till now I couldn't get a useful OTA trace.
Regarding the stack, as far as I have seen in targets.json the code for the NRF52_DK should use Cordio. Using nativ Nordic SDK code with the Softdevice everything seems to work.
But this is not really compatible with Mbed anymore as far as I have seen.

f43i4n, 2020-04-01:
I have recorded this issue with a sniffer on another NRF52 DK:

[wireshark trace](https://github.com/ARMmbed/mbed-os/files/4413990/mbed_issue_12191.pcapng.zip)

The issue seems to be related to switching to LE 2M PHY. This issue does not occur if I deactivate LE 2M PHY.

Some information:
Central: iPhone 11 Pro running iOS 13.3.1 using Nordic NRF connect app
Peripheral: NRF52 DK (PCA10040) running BLE_LED example from current `master` branch

sr425, 2020-04-18:
@desmond-blue Does the provided information help with figuring out what could be the issue?

desmond-blue, 2020-04-23:
I don't have NRF52_DK in hand so I can only guess...
1. From the trace, I can see the No. 39 packet which is an ack on 2M PHY from Slave, it seems 2M phy is configured well.
2. However there is no ack from Slave after No. 40 which is ATT Exchange MTU Request, it seems link layer didn't receive the packet, so the ball goes back to PHY.
3. Here we need another trace @f43i4n  without enabling 2M PHY.
4. And an experiment, could you @f43i4n  @sr425 modify [this line](https://github.com/ARMmbed/mbed-os/blob/ecf40080469e653bc0eb79b7f19420a238e5bc55/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/stack/sources/pal_bb_ble.c#L451) to
` *(volatile uint32_t *) 0x40001774 = ((*(volatile uint32_t *) 0x40001774) & 0xfffffffe) | 0x01000000; ` ?
Which is according to [Nordic errata ](https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52832_Rev2%2FERR%2FnRF52832%2FRev2%2Flatest%2Ferr_832.html) 143 RADIO: False CRC failures on specific addresses.

eriklins, 2020-05-08:
Hello,
I was facing the same issue with iPhone 11 and a Laird BL652 BLE module (nRF52832 based) as device. As far as I could see, the iPhone 11 tried to do a phy change request to 2MPHY and CODED_PHY. The default behaviour on the BL652 (programmed with Laird's SmartBASIC) was to just accept any phy request - which does not make much sense in this particular case as the BL652 does not have coded phy. So I changed the response to the phy request so that I change my phy to 1MPHY only and not accept the change to 2MPHY or CODED_PHY. That way the iPhone 11 accepted this and did not disconnect due to timeout.

yogeshk19, 2020-05-21:
@eriklins  I was having the same issue, with IPhone 11, with a custom App i wrote, how do I only allow 1MPHY and not accept the phy request changes in Cordio? Is there a specific API in Cordio to reject the Phy update requests?

Thanks,
Yogesh

eriklins, 2020-05-22:
Hello Yogesh,

I program my BL654 (nRF53840 based) module in Laird's SmartBASIC 
language. Here I can catch the phy change request event and switch to 
only 1MPHY instead of accepting the iPhone 11 requested change to 2MPHY 
and CODED PHY. Since SmartBASIC is based on Nordic SDK, I think it 
should be possible to catch that phy change request as well, but I 
cannot give code examples for Nordic SDK.

Best regards,
ER!K

pan-, 2020-05-29:
@yogeshk19 Unfortunately, the controller manages the PHY being used. At the host level it is only possible to indicate the preferred PHY to use. 
Have you tried to use [Gap:: setPreferredPhys](https://github.com/ARMmbed/mbed-os/blob/c7759fe505d922fe00ee3128dc60110f66647815/features/FEATURE_BLE/ble/gap/Gap.h#L1183-L1185) ?

pan-, 2020-10-21:
@sr425 Is it still the case ? We added proper privacy support for Mbed OS and it should solve the issues related to instant timeout on iOS. Can you try it out with master ?

pentabarf, 2020-10-27:
> @yogeshk19 Unfortunately, the controller manages the PHY being used. At the host level it is only possible to indicate the preferred PHY to use.
> Have you tried to use [Gap:: [...]

Thank you this solved my problems
For Everyone with similar issues:
`ble::phy_set_t phys(/* 1M */ true, /* 2M */ false, /* coded */ false);
  ble_error_t error = this->ble->gap().setPreferredPhys(/* tx */ &phys, /* rx */ &phys);`

pan-, 2020-10-29:
@sr425 This issue should be addressed in the latest version of Mbed OS that improves interoperability through proper privacy support. Since I haven't heard back from you, I'm closing the issue, please reopen it if it is required .

closed 2020-10-29 by pan-