ARMmbed/mbed-os #12192: ESP8266 SNTP function

star297, 2020-01-06:
### Description of defect

SNTP function missing

#### Target(s) affected by this defect ?
All
#### Toolchain(s) (name and version) displaying this defect ?
Any
#### What version of Mbed-os are you using (tag or sha) ?
On-line + Studio

Can someone drop in the ESP8266 built-in SNTP function code?
I wouldn't win an award for the best code, but I've been using the code similar to below for some time now, very compact and no need for a NTP library.
There's no DST function but that can be done outside the library.

`bool ESP826::setESPNTP(uint32_t Tz)`
`{`
   ` //  "cn.ntp.org.cn","ntp.sjtu.edu.cn","us.pool.ntp.org"`    
   ` //  Tz=2 for Europe;`     
   ` sprintf(ATcmd,"AT+CIPSNTPCFG=1,%d",Tz);`
   ` SendCMD(ATcmd);`    
   `// some timeout checking here`   
    `return true;`
}

`bool ESP826::getESPNTP(uint32_t setRTC)`
`{`    
   ` SendCMD((char*)"AT+CIPSNTPTIME?");       // get ESP built in SNTP Time`
   ` sscanf(readESP(),"+CIPSNTPTIME:%s %s %d %d:%d:%d %d",da,mo,&dt,&th,&tm,&ts,&ye); //` `format Thu Aug 04 14:48:05 2016`   
    `struct tm s;`
   ` s.tm_sec = (ts);`
   ` s.tm_min = (tm);`
    `s.tm_hour = (th);`
    `s.tm_mday = (dt);`   
   ` s.tm_mon = 0;`
   ` if (strcmp(mo, "Feb") == 0) {s.tm_mon = 1;}`
   ` if (strcmp(mo, "Mar") == 0) {s.tm_mon = 2;}`
    `if (strcmp(mo, "Apr") == 0) {s.tm_mon = 3;}`
   ` if (strcmp(mo, "May") == 0) {s.tm_mon = 4;}`
   ` if (strcmp(mo, "Jun") == 0) {s.tm_mon = 5;}`
   ` if (strcmp(mo, "Jul") == 0) {s.tm_mon = 6;}`
   ` if (strcmp(mo, "Aug") == 0) {s.tm_mon = 7;}`
   ` if (strcmp(mo, "Sep") == 0) {s.tm_mon = 8;}`
   ` if (strcmp(mo, "Oct") == 0) {s.tm_mon = 9;}`
   ` if (strcmp(mo, "Nov") == 0) {s.tm_mon = 10;}`
   ` if (strcmp(mo, "Dec") == 0) {s.tm_mon = 11;}`       
   ` s.tm_year = (ye - 1900);`  
  `  if(ye==1970){`
      `  return false;`
      `  }`
       ` else{`
           ` if (setRTC){set_time(mktime(&s));}  // set RTC clock if valid time`
           ` return s;`
           ` };`
`}`

0xc0170, 2020-01-08:
cc @ARMmbed/mbed-os-ipcore

star297, 2020-01-22:
Closing issue as fixed on #12300

closed 2020-01-22 by star297

michalpasztamobica, 2020-01-22:
@star297 , let's see if it passes code review and gets merged ;-)