ARMmbed/mbed-os #12212: NUMAKER_M252KG blinky builds fail

VeliMattiLahtela, 2020-01-08:
### Description of defect

NUMAKER_M252KG builds have started to fail in blinky build.

#### Target(s) affected by this defect ?
NUMAKER_M252KG

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6, GCC, IAR

#### What version of Mbed-os are you using (tag or sha) ?

6b89d16dd3
mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
Try building Blinky for the target.

0xc0170, 2020-01-08:
I believe this is an offender #12176

cc @ARMmbed/team-nuvoton

cyliangtw, 2020-01-08:
The root cause is #12176 not sense M252KG merged into mainstream in the duration of developing stage. Now, to fix it in PR #12217, the usbd implementation will be only included as while DEVICE_USBDEVICE is enabled.

0xc0170, 2020-01-08:
PR merged, this is now fixed

closed 2020-01-08 by 0xc0170