ARMmbed/mbed-os #12219: target

JanneKiiskila, 2020-01-08:
### Description of defect

The target testing tool does not correctly recognize the lack of "sector" -keyword in the index.json -file. This is a problem, as the tests will fail with an error, which does not tell you WHERE the actual issue itself is.

It should be fixed to the Python script, so that the target name gets printed out correctly in the assert (probably one more asset needs to be added?).

We get this:

```
=================================== FAILURES ===================================
_____________________________ test_bl_has_sectors ______________________________
    def test_bl_has_sectors():
        """Assert a bootloader supporting pack has sector information"""
        cache = Cache(True, True)
        named_targets = (
            target for target in TARGETS if
            (hasattr(target, "device_name") and getattr(target, "bootloader_supported", False))
        )
        for target in named_targets:
            assert target.device_name in cache.index,\
                ("Target %s contains invalid device_name %s" %
                 (target.name, target.device_name))
>           assert cache.index[target.device_name]["sectors"],\
                ("Device name %s is misssing sector information" %
                 (target.device_name))
E           KeyError: 'sectors'
```

Reason most likely is that the whole "sectors" -value is NOT for that target in the `index.json` -file.
Originating offending PR is here; https://github.com/ARMmbed/mbed-os/pull/12163

#### Target(s) affected by this defect ?

Any target, which has a bootloader enabled but does not have the sectors information in the index.json, but good luck figuring out which one. :-)

#### Toolchain(s) (name and version) displaying this defect ?

Travis/CI - Python

#### What version of Mbed-os are you using (tag or sha) ?

Current master, 4b0c4a073c9e2920c9ec41ec84386b64f70fbc89

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

Update index.json and add a new entry without the "sectors" -information, most likely.

JanneKiiskila, 2020-01-08:
@0xc0170 @adbridge

0xc0170, 2020-02-24:
I'll close this as fixed via #12226

closed 2020-02-24 by 0xc0170