ARMmbed/mbed-os #12221: Division operation much slower with microlib than other libs on cortex-M0

LMESTM, 2020-01-08:
### Description of defect

Following investigations in #12116 , it turns out that division with microlib seems much slower than ARM lib (and possibly also other libs) on cores like cortex-M0.

Few measurement on a "divided by 1000" operation were measured and are reported here for background information: 

Board | Toolchain | Division | Operation time (incl. IOs toggling) | Frequency during wake-up
-- | -- | -- | -- | --
L476RG | uARM | Native ( / ) | 440ns | 48MHz
L476RG | uARM | Div1000 | 480ns | 48MHz
L476RG | ARMC6 | Native ( / ) | 440ns | 48MHz
L476RG | ARMC6 | Div1000 | 480ns | 48MHz
L073RZ | uARM | Native ( / ) | 86µs | 4MHz
L073RZ | uARM | Div1000 | 226µs | 4MHz
L073RZ | uARM | Div10 (3 times) | 18µs | 4MHz
L073RZ | ARMC6 | Native ( / ) | 11,6µs | 4MHz
L073RZ | ARMC6 | Div1000 | 13µs | 4MHz
L073RZ | ARMC6 | Div10 (3 times) | 18µs | 4MHz

#### Target(s) affected by this defect ?

ALL

#### Toolchain(s) (name and version) displaying this defect ?

microlib toolchain (µARM)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os master of Jan 2020.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant here.

#### How is this defect reproduced ? 

See detailed history in #12116

0xc0170, 2020-01-09:
What would be suggestion here? From the link in the #12116, this is known issue with microlib. 

It might be worth reaching out internal team who is in charge for microlib ?

kjbracey-arm, 2020-01-09:
For division, microlib is doing what's being advertised, which is being smaller and slower. And I'm inclined to think the trade-off is okay. It's 40 bytes instead of 340 bytes. 8 times slower (in this case), 8 times smaller. 8 times slower isn't enough to be *that* much of a problem, except when running at like 4MHz.

I'm more concerned about the 64-bit multiply (which the `div1000` was using there). That is tens of times slower, AND many times bigger. It would be better in all regards to just use the standard library routine. We should raise this with the tools team.

MarceloSalazar, 2020-02-05:
Note we're waiting from Arm to comment and confirm how to proceed.

MarceloSalazar, 2020-02-19:
Note this affects all Cortex-M0/M0+.
@kjbracey-arm can you please review?

MarceloSalazar, 2020-03-18:
@kjbracey-arm any update on this?

evedon, 2020-04-29:
This issue seems to also affect cortex M4 (STM32F303 for example). See #12880

kjbracey-arm, 2020-09-10:
A fix for this is now expected in a future ARM Compiler 6 update - the compiler team has implemented new multiply and divide routines.

adbridge, 2020-10-15:
@LMESTM as this appears to be more of a compiler issue can we close this now ?

0xc0170, 2020-10-19:
The latest release of Arm Compiler is 6.15 (this versin contains the fix). I reviewed the internal tickets, it should be in the release.

I'll close this now as resolved.

kjbracey-arm, 2020-10-19:
AFAIK, you will need to actually activate the fast division routines, by adding some sort of import directive somewhere. The libraries will default to slow.

(And there might be toolchain-backward-compatibility issues with that. Maybe you can make it a weak reference to not break old toolchains?)

0xc0170, 2020-10-26:
We will need to review the latest release and how to make them enabled.

closed 2021-10-20 by ciarmcom