ARMmbed/mbed-os #12229: NRF52811 support

vvzvlad, 2020-01-09:
I have a developed device with NRF52811 that I would like to run under mbed OS. As far as I can see, there is no support for 52811 in mbed now.
What do I have to do to get it to appear? What prevents it from appearing?
I plan to use only 802.15.4 radio without BLE AoA, AoD and LR.

The zephyr project has support for this processor (https://docs.zephyrproject.org/latest/boards/arm/nrf52811_pca10056/doc/index.html), and it seems to me that adding support to mbed will not be very difficult: https://github.com/zephyrproject-rtos/zephyr/tree/master/soc/arm/nordic_nrf/nrf52

Links
https://www.nordicsemi.com/Products/Low-power-short-range-wireless/nRF52811

loverdeg-ep, 2020-01-09:
Nordic Semi for whatever reason doesn't support Mbed anymore.

The only reason NRF52840 and others still seem to have support is because ARM, Embedded Planet, and other partners/community members have done Nordic's responsibility for them.

Support for the NRF52811 should be straightforward but I'm not sure how well it will be maintained once implemented.

loverdeg-ep, 2020-01-09:
Also this is a question better left for forums.mbed.com based on the new community policies.

0xc0170, 2020-01-10:
> Also this is a question better left for forums.mbed.com based on the new community policies.

Correct. I'll close this , please move this to the forum. Thanks!

closed 2020-01-10 by 0xc0170