ARMmbed/mbed-os #12232: gcc: 'register' storage class specifier is deprecated and incompatible with C++17

JojoS62, 2020-01-09:
### Description of defect

#### Toolchain(s) (name and version) displaying this defect ?

gcc is complaining about using the register storage class as it is deprecated with c++11.
This issue has been reported for ARMC6 and been fixed there by suppressing the warning in 
#10717 
The same fix can be applied to the settings for gcc by using '-Wno-register', then it will compile also with -gnu++2a without warnings.

#### What version of Mbed-os are you using (tag or sha) ?
 mbed-os-5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
gcc 9.x

#### How is this defect reproduced ? 
compile with custom profile and -std=g++17 or -std-g++2a enabled.

loverdeg-ep, 2020-01-09:
@kjbracey-arm

kjbracey-arm, 2020-01-10:
Feel free to contribute a patch doing this - I'd have no objection.

Although we should be cleaning up the codebase. It's deprecated in C++14 and not actually in C++17 at all, so GCC would just be supporting it as an extension in that mode.

jeromecoutant, 2020-01-10:
Hi

I agree I am very late to align ST FW driver in mbed with ST cube delivery
https://github.com/STMicroelectronics/STM32CubeF4

Work is on going.

0xc0170, 2020-01-10:
@JojoS62 Can you specify which files are generating this ? To know who can help fixing it in case there are 3rd party drivers involved (as @jeromecoutant mentioned ST cube).

kjbracey-arm, 2020-01-10:
It will actually be almost exclusively 3rd party HAL code.

JojoS62, 2020-01-10:
yes, turning off the warning is just curing the symptoms, not really fixing the original problem.
in more recent CMSIS than in Mbed, this is fixed also in 2018 already:
https://github.com/ARM-software/CMSIS_5/issues/345
https://github.com/ARM-software/CMSIS_5/commit/f46ec37262bc4fd82b6c91d24198cc3b4a499fe3

I was told the deprication for 'register' was announced in 2009 already and is effective with c++11. But gcc or other compiler started to complain with c++17 setting.

0xc0170, 2020-01-10:
@JojoS62 We updated files in 2019 (the references are from 2018). I believe this should be fixed in CMSIS files in our code base.

kjbracey-arm, 2020-01-10:
> I was told the deprication for 'register' was announced in 2009 already and is effective with c++11. But gcc or other compiler started to complain with c++17 setting.

You're right - I thought it was deprecated in C++14, but it was deprecated in C++11. It's removed in C++17, but it doesn't stop the compilers still accepting it - they're just required to start complaining because it's not valid C++.

JojoS62, 2020-01-10:
> @JojoS62 Can you specify which files are generating this ?

  f.e. https://github.com/STMicroelectronics/STM32CubeF4/blob/b5abca20c9676b04f8d2885a668a9b653ee65705/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L1828
just a quick check, its still in the source in the link from Jerome.

JojoS62, 2020-01-10:
> they're just required to start complaining because it's not valid C++17.

yes, but very haevy... And I think the compiler warning should remain, it makes sense to show that some part of the code is depracted.

kjbracey-arm, 2020-01-10:
Problem is that if it's like ARMC6 it won't just warn once, or once per file, it will be 100s of times in a header file included by every C file...

Just to clarify, it gnu++17 normally accepts it with a warning, but `-Wno-register` disables the warning, right?

JojoS62, 2020-01-10:
yes, thats correct. And that is how it was fixed for ARMC6, only the compile switch has different name. For ARMC6, it has been turned off already: "-Wno-deprecated-register" is in the profiles.

JojoS62, 2020-01-10:
There are thousands of warnings because 'register' is used in many headerfiles.

I'm currently compiling, it looks like affected headers are _ll_rtc.h, _ll_tim.h and _ll_adc.h.

kjbracey-arm, 2020-01-10:
> For ARMC6, it has been turned off already

Because ARMC6 produces the warning for C++14 mode, so it needs the warning suppressed in the C++14 profile.

If you're making your own custom C++17 profile for GCC then you can obviously add the warning suppression in the same custom profile. But if you want to submit a patch to put it in the base profile, that's fine. I guess it makes it easier to switch to C++17 mode (possibly through a menu in an IDE export).

JojoS62, 2020-01-10:
yes, I can prepare the PR. But again, this is curing the symptoms. The ideal solution would be a clean HAL code. Removing the 'register' directives from the code makes no change in the compilation as the compilers ignore this for many years already.

JojoS62, 2020-01-10:
> If you're making your own custom C++17 profile for GCC then you can obviously add the warning suppression in the same custom profile

sure, this is possible. But the deprecated code is not a problem with Mbed, is there a chance that the HAL gets updated? Can I make a PR in the HAL repo?

0xc0170, 2020-01-10:
@JojoS62 Yes you can make PR updating HAL.

I would like to triage this issue but as it is too generic, I am not certain where it belongs and who should fix it (teams). We should split this one to modules. We can help but first should find offenders there. Besides ST driver, is there any other?

0xc0170, 2020-02-06:
@JojoS62 Lets simplify this issue. It won't move anywhere with the current scope

Besides ST driver, if there are any other, please create separate issues. I'll close this one.

closed 2020-02-06 by 0xc0170