ARMmbed/mbed-os #12247: DISCO_L475VG - Socket Address based get_address() not working

JanneKiiskila, 2020-01-13:
Please see - https://github.com/ARMmbed/mbed-os-example-pelion/pull/89 

Disco board does not pass CI verification:
```
2020-01-13T14:58:21.490Z | D1 <<< DutThread: get_ip_address failed with -3002
```

https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/nsapi_types.h#L40

Bug in the network driver? IP Address resolving must be a supported feature.

### Description of defect

Following code snippet failing with DISCO_L475VG.
String based `get_ip_address()` works.

```
    status = network->get_ip_address(&sa);
    printf("Network initialized, connected with IP %s\n\n", network->get_ip_address());	    if (status!=0) {
        printf("get_ip_address failed with %d\n", status);
        return -2;
```

#### Target(s) affected by this defect ?

DISCO_L475VG

#### Toolchain(s) (name and version) displaying this defect ?

Both ARM GCC 2019 and ARM CC 6.13.

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 5.15.0 official release (RC3).

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Use mbed-os-example-pelion PR https://github.com/ARMmbed/mbed-os-example-pelion/pull/89

michalpasztamobica, 2020-01-14:
Please try updating the version of `wifi-ism43362` driver [in here](https://github.com/ARMmbed/wifi-ism43362/commit/1978369b2310ea3955715b67869b717fd224a74c). Socket-based API was added to the driver on 26 Nov with [this commit](https://github.com/ARMmbed/wifi-ism43362/commit/534e644e976a6ce27b6e6aa602451fde31791cf2).

JanneKiiskila, 2020-01-14:
Excellent, thank you @michalpasztamobica !

JanneKiiskila, 2020-01-14:
One issue resolved, but seems the Wi-Fi driver itself has some connectivity issues. I'm failing to get it to work with my phone for example (Nokia 8.1 with Qualcomm chipset - has worked flawlessly with other Wi-Fi options with Mbed OS so far).

```
Failed to connect! error=-3011. Retry 1/5                                       
Failed to connect! error=-3011. Retry 2/5                                       
Failed to connect! error=-3011. Retry 3/5                                       
Failed to connect! error=-3011. Retry 4/5                                       
Failed to connect! error=-3011. Retry 5/5 
```

With a bigger Wi-Fi router we get this;

```
Connecting with interface: Wi-Fi to mbed                                        
Failed to connect! error=-3011. Retry 1/5                                       
IP: 192.168.100.16                                                              
Network initialized, registering...                                             
                                                                                
Error occurred : MbedCloudClient::ConnectDnsResolvingFailed                     
Error code : 12                                                                 
Error details : Client in reconnection mode DnsResolvingFailed  
```

So, it will actually authenticate in (why not on the 1st time already), but DNS resolving fails anyway...

michalpasztamobica, 2020-01-14:
@JanneKiiskila , could it be too many wifi networks and range + old firmware as reported in https://github.com/ARMmbed/wifi-ism43362/issues/68 (originally reported in https://github.com/ARMmbed/mbed-os-example-wifi/issues/177 and most information is here)?

JanneKiiskila, 2020-01-14:
Sounds like I need to update the Wi-Fii module firmware then.

michalpasztamobica, 2020-01-14:
Maybe @JuhPuur can help with this? Or maybe there is already an updated board available for you?
Also - if you just want to run once to see proper results, you could try @korjaa 's solution with wrapping your board with this foil (mentioned at the end of mbed-os-example-wifi issue) to limit the scan results.

korjaa, 2020-01-14:
The board I gave to Janne should have the latest WiFi firmware. But of course the first rule of debugging is to not believe anything the previous person says.

JanneKiiskila, 2020-01-14:
Any easy way to check it?

korjaa, 2020-01-14:
It does print some version info if you enable AT debugs. But how were those enabled again I don't remember out of my head.

michalpasztamobica, 2020-01-14:
Yes, try flashing the binary attached to https://jira.arm.com/browse/ISGTESTINF-4357 (or just enable the ism43362 logs in your binary). The initial printout from the driver is something like:
```
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3525
```
version `3523` is old, version `3525` is new.

korjaa, 2020-01-14:
Could it be cool if all this version checking happened automatically in the driver :P

michalpasztamobica, 2020-01-14:
To enable the ism4336 logs, the mbed_app.json should be something like this:
```
        "DISCO_L475VG_IOT01A": {
            "nsapi.default-wifi-security": "WPA_WPA2",
            "ism43362.provide-default": true,
            "wifi-debug": true
        },

```
and don't forget to add `-v` to `mbed run`/`mbedgt` otherwise you don't get the printouts.

JanneKiiskila, 2020-01-14:
Seems it gets printed out by default with mbed-os-example-pelion, 
```
ISM43362Interface: read_version = 3525
```

gets printed out, so it is the newest. Just out of curiosity I used the Wi-Fi scanner and checked how many Wi-Fi access points we have here and well... Let's just say it's probably 10x more than I anticipated.

michalpasztamobica, 2020-01-14:
To exclude the environment's influence - could you try to flash your binary to raas? mervi has a few DISCOs with `RF-BOX` tag. They are living in a Wi-Fi paradise, with just those networks available that we use for tests (make sure you adjust credentials). 
The nightly runs don't indicate any issues in network tests with this platform.

JanneKiiskila, 2020-01-14:
Trials continue at home, where there is a LOT less Wi-Fi activity. However, situation does not get better towards the phone - it simply does not even find that as in the scan.

```
WiFi example                                                                    
Mbed OS version 5.15.0                                                          
                                                                                
Scan:                                                                           
Network: KIISKILA-NET secured: WPA2 BSSID: A0:40:A0:7c:ca:f1 RSSI: -47 Ch: 1    
Network: HP-Setup>e3-M277 LaserJet secured: None BSSID: 6A:14:1:26:f0:e3 RSSI: 6
Network: KIISKILA-NET-DNA secured: WPA2 BSSID: D8:D7:75:68:aa:d6 RSSI: -70 Ch: 1
3 networks available.                                                           
                                                                                
Connecting to SSID...    
Connection error: -3011 
```
So, no luck there. There is **some real issue here** (using the master of [mbed-os-example-wifi](https://github.com/ARMmbed/mbed-os-example-wifi) here to rule out any client SW dependency).

When I then use the cable  modem Wi-Fi, it manages to connect to that.

```
WiFi example                                                                    
Mbed OS version 5.15.0                                                          
                                                                                
Scan:                                                                           
Network: KIISKILA-NET secured: WPA2 BSSID: A0:40:A0:7c:ca:f1 RSSI: -47 Ch: 1    
Network: HP-Setup>e3-M277 LaserJet secured: None BSSID: 6A:14:1:26:f0:e3 RSSI: 6
Network: Omni_3D3A00 secured: WPA2 BSSID: 34:21:9:3d:3a:11 RSSI: -93 Ch: 6      
Network: KIISKILA-NET-DNA secured: WPA2 BSSID: D8:D7:75:68:aa:d6 RSSI: -69 Ch: 1
4 networks available.                                                           
                                                                                
Connecting to KIISKILA-NET...                                                   
Success                                                                         
                                                                                
MAC: C4:7F:51:05:42:2B                                                          
IP: 10.0.0.157                                                                  
Netmask: 255.255.255.0                                                          
Gateway: 10.0.0.1                                                               
RSSI: -46                                                                       
```

JanneKiiskila, 2020-01-14:
With mbed-os-example-wifi - if I add the address resolving for www.arm.com, it works. 
However, with the mbed-cloud-client-example, it will just fail.

```
ISM43362Interface: read_version = 3525                                          
Start Device Management Client                                                  
Using hardcoded Root of Trust, not suitable for production use.                 
Starting developer flow                                                         
Developer credentials already exist, continuing..                               
Application ready. Build at: Jan 14 2020 23:02:20                               
Mbed OS version 99.99.99                                                        
mcc_platform_interface_connect()                                                
ISM43362Interface: attach                                                       
Connecting with interface: Wi-Fi to KIISKILA-NET                                
        ISM43362: connection_status 3                                           
ISM43362Interface: update_conn_state_cb 2 -> 3                                  
        ISM43362: connection_status 1                                           
ISM43362Interface: update_conn_state_cb 3 -> 1                                  
ISM43362Interface: connect_status 0                                             
        ISM43362: receivedIPAddress: 10.0.0.157                                 
        ISM43362: receivedIPAddress: 10.0.0.157                                 
IP: 10.0.0.157                                                                  
Network initialized, registering...                                             
ISM43362Interface: attach                                                       
        ISM43362: receivedIPAddress: 10.0.0.157                                 
ISM43362Interface: socket_open id=0 proto=1                                     
ISM43362Interface: socket_attach id 0                                           
        ISM43362: open ok with id 0 type 1 addr 8.8.8.8 port 53                 
ISM43362Interface socket_send_nolock id 0 size 51                               
        ISM43362 send: id 0 amount 51                                           
        ISM43362 check_recv_status: id 0                                        
        ISM43362 check_recv_status: id 0 read_amount=0                          
        ISM43362 check_recv_status: id 0                                        
        ISM43362 check_recv_status: id 0 read_amount=0                          
        ISM43362 check_recv_status: id 0                                        
        ISM43362 check_recv_status: id 0 read_amount=109                        
ISM43362Interface socket_check_read read_amount 109                             
ISM43362Interface socket_recv: recv=109                                         
        ISM43362 check_recv_status: id 0                                        
        ISM43362 check_recv_status: id 0 read_amount=0                          
ISM43362Interface: socket_attach id 0                                           
ISM43362Interface: socket_close, id=0                                           
        ISM43362: CLOSE socket id=0                                             
                                                                                
Error occurred : MbedCloudClient::ConnectDnsResolvingFailed                     
Error code : 12                                                                 
Error details : Client in reconnection mode DnsResolvingFailed   
```

Based on the logs, it will get 109 bytes of data from 8.8.8.8 (which I think is Google's DNS service), so I think it should have received the address? So, something funny is now happening here.

michalpasztamobica, 2020-01-15:
@JanneKiiskila, to make sure I understand right, there are now two failure modes right now:
1) DISCO_L475VG_IOT01A with the ISM43362 wifi chip fails to connect to an AP set on your smartphone (other platforms connect just fine).
1) mbed-cloud-client-example fails to resolve DNS address even after connecting to a working router?

@LMESTM , @jeromecoutant would you please take a look at this issue? Perhaps this is something you have heard of? Please bear in mind - the current issue is different from the one originally reported and present in description. The issue only occurs with this one platform (and the connection issue occurs just with one particular AP).

JanneKiiskila, 2020-01-15:
Well, turns out the phone AP thing is now regression in the phone side, even u-Blox ODIN can't connect to it anymore. Seems one of the recent updates to the phone have broken it.

michalpasztamobica, 2020-01-15:
So sounds like issue (1) - not being able to connect to phone's AP - is not relevant any more, but issue (2) - DNS resolution failure - is still there? And just for DISCO_L475VG_IOT01A?

JanneKiiskila, 2020-01-15:
Issue 2) is also only now an issue with mbed-cloud-client-example. The mbed-os-example-pelion does seem to work now (at least in a good enough Wi-Fi coverage). It didn't pass in the office environment, but I managed to get it connected at home.

JanneKiiskila, 2020-01-15:
The DNS issue also happens with u-blox, so it now seems it's more of an issue (memory?) when using the slightly more complicated mbed-cloud-client-example vs. mbed-os-example-pelion. So, closing this issue out and investigations continue internally.

closed 2020-01-15 by JanneKiiskila