ARMmbed/mbed-os #12250: Is mbedOS 6 going to support LittleFS2?

pilotak, 2020-01-14:
### Description of defect
Since OS 5 has LittleFS v1, is OS 6 going to support LittleFS2 by default? as a major release it could it's a good opportunity - LFS2 has been rewritten and has relatively continuous buf fixes. I know there is a [repo](https://github.com/ARMmbed/mbed-littlefs) trying to bring those changes in but is not up to date and unfortunately since OS5.15 has a new CRC engine, it’s not compatible anymore.
At the moment i'm unable to fix the CRC thing, other than that i can contribute. At the repo i submitted a PR for V2.1.1 and in issues you will find a migration script.

I already asked on forum but nobody responded
https://forums.mbed.com/t/is-mbedos-6-going-to-support-littlefs2/6694

#### Target(s) affected by this defect ?
all

#### Toolchain(s) (name and version) displaying this defect ?
all

#### What version of Mbed-os are you using (tag or sha) ?

5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-01-14:
Sorry for the delay in the forums, it's better place for this there as it is not actual bug report.

> At the moment i'm unable to fix the CRC thing, other than that i can contribute. At the repo i submitted a PR for V2.1.1 and in issues you will find a migration script.

I will mention the team who could answer it and help.
cc @ARMmbed/mbed-os-storage 

However, as this is not technical bug (feature request), forum is proper place, will get an answer there. I'll close this.

> At the moment i'm unable to fix the CRC thing, other than that i can contribute. At the repo i submitted a PR for V2.1.1 and in issues you will find a migration script.

This is the PR: https://github.com/ARMmbed/mbed-littlefs/pull/14 ?  There is no mention of the issue, no description provided.

0xc0170, 2020-01-14:
cc @NadyaARM

closed 2020-01-14 by 0xc0170

pilotak, 2020-02-11:
I have managed the bring v2.1.4 to up to date OS, fixed most of the test with travis, please could you have a look at it? https://github.com/ARMmbed/mbed-littlefs/pull/14

This is a great opportunity to bring LittleFS v2 to OS6

@NadyaARM @0xc0170

0xc0170, 2020-02-11:
@pilotak Thanks for sharing the effort !

cc @ARMmbed/mbed-os-storage