ARMmbed/mbed-os #12268: Compiler warning - s

JanneKiiskila, 2020-01-16:
### Description of defect

```
Compile [ 22.1%]: SerialBase.cpp
[Warning] SerialBase.h@357,27: 'mbed::SerialBase::_rx_pin' will be initialized after [-Wreorder]
[Warning] SerialBase.h@335,24:   'void (mbed::SerialBase::* mbed::SerialBase::_init_func)()' [-Wreorder]
[Warning] SerialBase.cpp@26,1:   when initialized here [-Wreorder]
[Warning] SerialBase.h@358,28: 'mbed::SerialBase::_static_pinmap' will be initialized after [-Wreorder]
[Warning] SerialBase.h@335,24:   'void (mbed::SerialBase::* mbed::SerialBase::_init_func)()' [-Wreorder]
[Warning] SerialBase.cpp@44,1:   when initialized here [-Wreorder]
```

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0-alpha1 (but essentially almost all of them, older ones)...

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Irrelevant.

#### How is this defect reproduced ? 

`mbed compile` any target, essentially, for example:

`mbed compile -t GCC_ARM -m K66F`

0xc0170, 2020-01-16:
cc @ARMmbed/mbed-os-core

evedon, 2020-01-16:
@mprse This warning has been introduced with the latest change on static pin-map. Please look into it when you get a chance.

mprse, 2020-01-16:
Will check this tommorow.

mprse, 2020-01-20:
The fix can be found here: PR https://github.com/ARMmbed/mbed-os/pull/12282.

0xc0170, 2020-02-06:
Fixed via #12282

closed 2020-02-06 by 0xc0170