ARMmbed/mbed-os #12277: Can create, but not use, established PPP session on SARA-R410M-02B modem

MrkCDGHub, 2020-01-17:
I am able to establish a PPP connection with the Sara-R4 modem, but attempts to
create and use a TCP socket connection, supplying an actual IP address to an echo test server always fail with -3004.
Attempting to use a hostname and DNS also fails, although I was successful once or twice by supplying a hostname
but could never reproduce it.

I am able to successfully Connect and transmit to test Servers (broker.hivemq.com or test.mosquitto.org)
using AT socket commands.

This seems to be the same exact problem described here: https://github.com/ARMmbed/mbed-os/issues/7464
with similar PPP problems reported for the Sara-R4.

Generally it looks like the recommended 'fix' for this problem has been to send AT&S0 before establishing the PPP 
connection, but I do that, and in fact I don't have any problem establishing the PPP connection, just trying to use it.

I have also tried this with a simpler application (mbed-os-example-cellular-master), modified for my target
with the same results.

Additionally, exiting PPP mode is never successful using the +++ or DTR (via GPIO control) methods described in the Sara-R4 AT command manual
Per the SARA-R4 AT command documentation:
The data session is terminated by one of the following events:
• sending ~+++.
• via a DTR transition from ON to OFF.
• sending an LCP Terminate Request.

My target board uses an MTQN-MNG1-B01, modem firmware as follows:
Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.08 [Apr 17 2019 19:34:02]
mbed-os-5.15 
[PPPFails.log](https://github.com/ARMmbed/mbed-os/files/4077205/PPPFails.log)

0xc0170, 2020-02-06:
cc @ARMmbed/mbed-os-wan @ARMmbed/team-ublox

0xc0170, 2020-02-06:
@MrkCDGHub Can you leave the issue template, is there for a reason (we can use the data from sections, etc). Thanks

AnttiKauppila, 2020-02-06:
@ARMmbed/team-ublox Could you check this?

MrkCDGHub, 2020-02-06:
I have an update:
The reason I was unable to establish a connection was apparently because IPv6 is not routable through lwip, so it must be set to IPv4. 
I made the following modifications to my mbed_app.json file:
            "lwip.ipv4-enabled": true,
            "ppp.ipv4-enabled": true,
            "lwip.ipv6-enabled": false,
            "ppp.ipv6-enabled": false,

and now I am able to reliably create and use a PPP connection.

That said - I still cannot escape from PPP data mode back to command mode.
We would like to keep the PPP session established and then periodically transmit up to to our backend with information including the Signal Quality. That is only available by returning to command mode and sending an AT+CSQ.
The documentation is vague about how to do this.
I've tried sending +++. ~+++ to the modem. I've tried toggling DTR.
None of these options were successful.
I know there's a Mux feature, but I've heard nothing positive about it and I don't have endless time to try and get that working.

Has anyone used PPP on this modem and successfully exited data mode?

fahim-ublox, 2020-02-07:
Internal reference PGM_NO_MBD-322: Customer unable to escape from PPP data mode back to command mode. We will update you shortly

mudassar-ublox, 2020-02-14:
I have checked with [mbed-os-example-cellular](https://github.com/ARMmbed/mbed-os-example-cellular) and its working fine and we can exit from data mode and able to send commands. After successfull communicaion modem return code `NO CARRIER` after that modem starts reponding `AT `commands, logs are attached.

Additionally I saw same issue when using manual AT commands that modem doesn't exit data mode with ~+++ or +++, we are looking into it.

[C030_Application_logs.txt](https://github.com/ARMmbed/mbed-os/files/4204265/C030_Application_logs.txt)
[Modem_Sniffer_Logs.txt](https://github.com/ARMmbed/mbed-os/files/4204266/Modem_Sniffer_Logs.txt)

MrkCDGHub, 2020-02-14:
When you indicate "we can exit from data mode and able to send commands", I assume that you mean you have completed the data transfer and closed the PPP connection. You were not able to exit Data mode with the PPP connection still in place via some method other than ~+++ or +++ and send AT commands. Is that accurate?

MrkCDGHub, 2020-02-21:
Any progress/update on this issue (unable to escape PPP data mode)?

mudassar-ublox, 2020-02-24:
First check your escape character with `ATS2?` then check escape prompt delay with `ATS12?` command. See the description of S12 command in [at command manual](https://www.u-blox.com/en/docs/UBX-17003787). It states that you need to send escape character with in time period define with `ATS12` command.

It will work like, wait 1 second (`escape prompt delay`) before sending '+' (`escape character`) then send all escape character '+' within 1 second then wait another 1 second before sending other AT command.

MrkCDGHub, 2020-02-25:
I've verified my S2 and S12 setttings, both at the default of '+' and 50 (20ms). With a scope I've verified the intercharacter delay of my +++ sequence, it's consistent and I've tried various delays from 5 to 20 ms. Never exits Data mode. Same with dropping DTR.

mudassar-ublox, 2020-02-25:
If S12 value is 50 then according to manual its 1 sec.

charlottewai, 2020-08-04:
Hi @MrkCDGHub  have you solved this? I am having the same problem to escape PPP mode to send AT command.

MrkCDGHub, 2020-08-04:
No I never did get it to work. I gave up and instead used the modem protocol stack. Someone reportedly was able to get it to work on the same hardware using a non mbed implementation. I wasted countless hours looking into this, it was very frustrating...

adbridge, 2020-10-15:
Based on the last comments I think we can close this.