ARMmbed/mbed-os #12279: SPIFBlockDevice spi concurrency issue

AlefeFelipe, 2020-01-17:
### Description of defect
SPIFBlockDevice loses blockcount when another thread that use spi causes a interruption. This cause an error on the mbed-os and the program crash

#### Target(s) affected by this defect ?
Tested in the NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?
Tested in GCC_ARM, version 6.3.1

#### What version of Mbed-os are you using (tag or sha) ?
Tested in version 5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Tested in mbed-cli, version 1.10.1

#### How is this defect reproduced ? 
This can be reproduced using a shared spi bus and, when using a operation with the spif blockdevice, cause a interruption in the spi bus using the other device conectado to it.

#### What was the used solution ?
The solution found was insert lock and unlock in the SPIFBlockDevice's spi object operations
[spif_block_device_solution.zip](https://github.com/ARMmbed/mbed-os/files/4077631/spif_block_device_solution.zip)

felipeLeast, 2020-01-28:
I've had the same problem, this solved the problem for me!

kjbracey-arm, 2020-02-05:
Duplicate of https://github.com/ARMmbed/mbed-os/issues/11732

0xc0170, 2020-02-06:
If that is the case, please @AlefeFelipe update #11732. I'll close this

closed 2020-02-06 by 0xc0170