ARMmbed/mbed-os #12284: Python requirements fail to install on Ubuntu 18.04 LTS

nat45928, 2020-01-21:
### Description of defect
Python requirements are failing to install during `mbed deploy` on Ubuntu 18.04 with. Running pip standalone (to see output) results in:

`
> python3 -m pip install -r /mbed-os/requirements.txt
...
Collecting python-dotenv>=0.8.2 (from mbed-cloud-sdk<2.1,>=2.0.6->-r /mbed-os/requirements.txt (line 9))
  Downloading https://files.pythonhosted.org/packages/65/16/43f00b087c7f3f20a8d8dc16d6f6e2eff4e662afcb31ec4a31535a5be244/python-dotenv-0.10.5.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-57adlm73/python-dotenv/setup.py", line 13, in <module>
        long_description = read_files(['README.md', 'CHANGELOG.md'])
      File "/tmp/pip-build-57adlm73/python-dotenv/setup.py", line 9, in read_files
        data.append(f.read())
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 2589: ordinal not in range(128)
`

Digging into the failing package revealed this issue:

https://github.com/theskumar/python-dotenv/issues/227

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-02-06:
Thanks @nat45928 for the report. The referenced issue was closed, can you confirm this is fixed now?

0xc0170, 2020-02-11:
Closed as resolved

closed 2020-02-11 by 0xc0170