ARMmbed/mbed-os #12290: crash_log_parser.py does not work with python3

teetak01, 2020-01-21:
### Description of defect

```
Traceback (most recent call last):
  File "mbed-os/tools/debug_tools/crash_log_parser/crash_log_parser.py", line 199, in <module>
    elfhelper = ElfHelper(args.elffile, args.mapfile)
  File "mbed-os/tools/debug_tools/crash_log_parser/crash_log_parser.py", line 38, in __init__
    self.matches = _PTN.findall(op)
TypeError: cannot use a string pattern on a bytes-like object
```

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Python 3.6.8

#### How is this defect reproduced ?

teetak01, 2020-01-21:
@madchutney

0xc0170, 2020-02-06:
cc @ARMmbed/mbed-os-core you could help? It's been added by core team a while back

mark-edgeworth, 2020-02-12:
I've put up a fix for this under https://github.com/ARMmbed/mbed-os/pull/12419

commit 2ad6f45d (2020-02-13): Fix #12290: crash_log_parser on py3
commit 7634f368 (2020-02-13): Merge pull request #12419 from ARMmbed/fix-gh12290

Fix #12290: crash_log_parser on py3
closed 2020-02-13 by 0xc0170