ARMmbed/mbed-os #12298: Cypress PSoC 62 boards missing support for managed bootloader

maclobdell, 2020-01-21:
### Description of defect

When building applications that require managed bootloader (aka. mbed-bootloader) support such as https://github.com/ARMmbed/mbed-os-example-pelion, the application needs to build correctly, without error.  

However, as of Mbed OS 5.15.0, there seems to be a new build check that is causing an error.   

`Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.`

This issue was previously highlighted in https://github.com/ARMmbed/mbed-os/issues/10892, but was able to be resolved by adding to the configuration of the application itself.

The configuration that was previously working is listed at https://github.com/ARMmbed/mbed-os-example-pelion/pull/70/files

Now it is clear that support needs to be added to the Mbed OS ports for the affected boards.  

I will investigate adding a PR for this.

#### Target(s) affected by this defect ?
Cypress PSoC 6 Wi-Fi BT Prototyping Kit
* https://os.mbed.com/platforms/CY8CPROTO-062-4343W/
PSoC 6 WiFi-BT Pioneer Kit
* https://os.mbed.com/platforms/CY8CKIT-062-WiFi-BT/

#### Toolchain(s) (name and version) displaying this defect ?
tested with ARM compiler, but should be independent.

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 

mbed import https://github.com/ARMmbed/mbed-os-example-pelion/pull/70/files

add configuration for CY8CPROTO_062_4343W to mbed_app.json.  (See https://github.com/ARMmbed/mbed-os-example-pelion/pull/70/files)

mbed compile -m  CY8CPROTO_062_4343W -t ARM

0xc0170, 2020-02-04:
cc @ARMmbed/team-cypress Please review

kyle-cypress, 2020-02-05:
If I understand the discussion in #10892 correctly, the missing values in question are mbed_rom_start and mbed_rom_size. @maclobdell , can you please confirm?
If so, these values were removed in bf00ebe4090144ee2f34ba2067f65fcb6317b5f9 by @hennadiykytsun . Hennadiy, can you please comment on what you think the right thing is to do here?

hennadiykytsun, 2020-02-06:
Previously it was the issue with the additional area allocation for the CM0+ image in the CM4 image, even if CM0+ component not selected. 
We have fixed this issue in the linker scripts. 
So, if someone going to customize ROM/RAM/APP size from the targets.json- he is responsible to customize the all following macros:
MBED_APP_START 
MBED_APP_SIZE
MBED_ROM_START
MBED_ROM_SIZE
MBED_RAM_START
MBED_RAM_SIZE
In this case linker will use provided from json addresses where the app will be applied

0xc0170, 2020-02-24:
@maclobdell Please review. Is this issue still valid?

maclobdell, 2020-03-24:
This is still an issue. I have had a conference call with Cypress to review it.

vmedcy, 2020-03-24:
Hello! I believe the bootloader support issue is solved by https://github.com/ARMmbed/mbed-os/pull/12679 - there exists non-trivial dependency between CMSIS export (tools/arm_pack_manager/index.json) and bootloader related code (sectors and regions properties in tools/config/__init__.py). I have not reviewed that python code in details to understand the exact relation/impact. I also don't know if any documentation exists on this topic.

0xc0170, 2020-04-02:
@vmedcy If any questions, I can try to look into this one as well and help :)

vmedcy, 2020-04-02:
@0xc0170 I submitted https://github.com/ARMmbed/mbed-os/pull/12752 with targets.json fix for mbed_ram_start.

loverdeg, 2020-07-24:
@0xc0170 Seeing a similar issue when trying to build the bootloader example for a cy62 target.