ARMmbed/mbed-os #12309: STM32L4R5ZI Tickless/Power Managment doesn´t work

Zeratuls, 2020-01-24:
### Description of defect
STM32L4R5ZI Tickless does not work properly. Using the debug profile the code works correctly but without going to sleep what is normal. In the case of develop and release profiles when we use ThisThread :: sleep_for (...) or EventFlags the device never wakes up.

#### Target(s) affected by this defect ?
NUCLEO -L4R5ZI

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli-1.10.2

#### How is this defect reproduced ? 
Using the release/develop profile any attempt to use the tickless in the code.

~~~~
#include "mbed.h"

Serial pc(USBTX, USBRX,115200);

int main()
{
    pc.printf("Start\r\n");
    while(1)
    {
        pc.printf("Loop\r\n");
        mbed_stats_cpu_t stats;
        mbed_stats_cpu_get(&stats);
        pc.printf("Uptime: %llu \r\n", stats.uptime / 1000);
        pc.printf("Idle: %llu \r\n", stats.idle_time / 1000);
        pc.printf("Sleep: %llu \r\n", stats.sleep_time / 1000);
        pc.printf("Deep Sleep: %llu \r\n", stats.deep_sleep_time / 1000);

        ThisThread::sleep_for(2000);        
    }
}
~~~~

LMESTM, 2020-01-24:
@Zeratuls just wondering whether this could be related to CPU stats. Could you try and remove the stats usage (and corresponding STAT configuration) and let us know if the same happens ?

LMESTM, 2020-01-24:
@jeromecoutant 
Also would you try removing Serial pc(USBTX, USBRX,115200);
and simply call to printf(); ?
That would be at 9600 bauds by default. Or you may increase the baudrate with mbed_app.json 
the objective is to see if the Serial instance is not causing troubles ...

Zeratuls, 2020-01-24:
@LMESTM I have removed the configuration for the STATs in mbed_app.json and I have used the Serial by default (9600 baud). Code:
~~~~
#include "mbed.h"

int main()
{
    while(1)
    {
        printf("Loop\r\n");
        ThisThread::sleep_for(2000);
    }
}
~~~~
The result is the same, it never wakes up.

Zeratuls, 2020-01-28:
@LMESTM Indeed, it seems that the Serial is responsible. This way it works:
```c++

#include "mbed.h"

DigitalOut led(LED1);

int main()
{
    while(1)
    {
        led = 0; // LED is ON
        //printf("LED 0");
        ThisThread::sleep_for(1500);
        led = 1; // LED is OFF
        //printf("LED 1");
        ThisThread::sleep_for(1500);
    }
}
```
But with printf it doesn't work. What solution do you propose?

Thanks

LMESTM, 2020-01-28:
@Zeratuls thanks for the update - this is a good hint. This will require further debugging  to understand what happens ... we're trying to have someone in the team have a look at it.

LMESTM, 2020-02-11:
@Zeratuls sorry for the delay ...

Not much progress here, neverthless let me share one observation. The board doesn't crash, it seems that there is "only" an issue with Serial configuration after deep sleep. The below code demonstrates it because the prints don't work after deep sleep, but the LED blinking continues ...

```
#include "mbed.h"

Serial pc(USBTX, USBRX,115200);
DigitalOut led(LED1);

int main()
{
    pc.printf("Start\r\n");
    led = 1;
    while(1)
    {
        led = !led;
        pc.printf("Loop\r\n");
        ThisThread::sleep_for(2000);        
    }
}
```

LMESTM, 2020-02-11:
@Zeratuls 
Update: proposed fix delivered in #12409 
For sake of clarity in the future or for other users, could you update your Issue title with something like "STM32L4R5ZI printf doesn't work with Tickless/PowerManagement"

closed 2020-02-13 by 0xc0170

Zeratuls, 2020-02-13:
@LMESTM 
Thanks for the fix!!!