ARMmbed/mbed-os #12313: ThisThread::sleep_for() issue in bare mode

mprse, 2020-01-27:
#### Description of defect

This issue is a result of investigation failing `tests-mbed_drivers-lp_ticker` and `tests-mbed_hal-rtc` tests in bare metal mode.

Please check the following code which shows the problem:

```
void test()
{
    Timer us_timer;
    //mbed::internal::init_os_timer();   // <----- Initialize OsTimer on start. Uncomment this to fix the problem.

    LowPowerTimer lp_ticker; // This will initialize the lp-ticker. lp-ticker starts counting.

    us_timer.start();
    while(us_timer.read_ms() < 30); // Let the lp-ticker to count (30 ms).
    us_timer.stop();

    us_timer.reset();
    us_timer.start();
    ThisThread::sleep_for(100); // Perform delay (100 ms). This will initialize OsTimer (~30 ms already passed) and use it to perform the delay.
    us_timer.stop();

    printf("sleep_for delay: %i \r\n", us_timer.read_ms()); // Print the real delay. This will give ~70 ms (instead 100 ms).

}

```

Timing tests fail because `OsTimer` (which uses `lp-ticker`) has not been started together with `lp-ticker` in bare metal mode. In the above example every call `ThisThread::sleep_for(100);` gives the `~70 ms` real delay.

The reason for this problem is time inconsistency between `osTimer(SysTimer)` and `lp-ticker`. `osTimer` operates on ticks. One tick corresponds to `1 ms`. `osTimer` start operationg when `lp-ticker` already counted `30 ms` and this corresponds to `osTimer` tick 0.
Please check the function used to perform delay in bare metal mode:

https://github.com/ARMmbed/mbed-os/blob/f469f710bca16abd1c538e48a3bee9db0a8e47e8/platform/source/mbed_os_timer.cpp#L230-L247

Wake time is calculated using `init_os_timer()->update_and_get_tick()` function and given delay in `ms`. In our case `init_os_timer()->update_and_get_tick()` returns 0 (just started), so the wake up time is: 
`0 ms (current tick)  + 100 ms  (delay) = 100 ms`.
Please check the function which uses wake up time to set the `lp-ticker` interrupt (`sleep_prepare(), os_timer->set_wake_time(wake_time)`):

https://github.com/ARMmbed/mbed-os/blob/f469f710bca16abd1c538e48a3bee9db0a8e47e8/platform/source/SysTimer.cpp#L88-L140

Here we can see that our wake up time (in `ms/ticks`) was converted to `us` (line 107) and `lp-ticker` interrupt was scheduled by means of `insert_absolute(wake_time)`. In our case interrupt is set to absolute time `100 000 us (100 ms)`, but `lp-ticker` current time is `~30 000 us (30 ms)`. That is why we are getting `70 ms` delay instead of `100 ms`.

I can see a few possibilities to solve this problem and I would like to discuss how this should be fixed before creating PR.

1. Start the `OsTimer` at startup (before `lp-ticker` is used).  This is the easiest solution, but I'm not sure if it is acceptable in the bare metal mode. I assume that in this mode we want to disable as many resources as possible, so if someone doesn't want to use `lp-ticker`, `ThisThread::sleep_for()`, etc. there is no need to have `lp-ticker and osTimer` started.

2. Change the implementation of `SysTimer` class:
https://github.com/ARMmbed/mbed-os/blob/f469f710bca16abd1c538e48a3bee9db0a8e47e8/platform/source/SysTimer.cpp#L69

Changing to `_time_us(0)` solves the problem, but I'm not sure if this is acceptable. This way we will have sys-tick 10 after init if `lp-ticker` has been already counting for 10 ms.  

3. Update the implementation of `ThisThread::sleep_for()`. In `do_timed_sleep_relative_or_forever()` function we can store the time offset between `lp-ticker` and `sys-timer`.

I would go with the second option.

Whats your opinion?
cc @jamesbeyond @0xc0170 @kjbracey-arm 
@maciejbocianski @fkjagodzinski 

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

706625a982

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

Run the provided code snippet.

jamesbeyond, 2020-01-27:
Hi @mprse, thanks for the very detailed explanation and investigation, this is really helpful.
Regarding to solution1, is this changes able to be made just for there 2 tests? or you mean `Start the OsTimer at startup` for all bare-metal mode?

mprse, 2020-01-27:
@jamesbeyond The issue is related to bare metal profile. The tests only show the problem. So the first solution is to start OsTimer at startup for entire bare metal mode.

jamesbeyond, 2020-01-28:
ping @evedon, this is the investigation results for the Baremetal failed test on K64F

evedon, 2020-01-28:
I don't think it is unreasonnable to start the OsTimer at startup in bare metal mode. The comment here https://github.com/ARMmbed/mbed-os/blob/master/platform/source/mbed_os_timer.cpp/#L28-LL33 suggests that a system global SysTimer is available for timed sleeps or absolute system timing. I'd rather that the system behaviour in RTOS or non-RTOS mode is similar.

@kjbracey-arm What are your thoughts?

kjbracey-arm, 2020-01-29:
Part of the point of bare metal is to not pay for stuff you're not using.

Starting the timer implicitly before main means (a) pulling in timer code into every image, and (b) paying the power cost for that.

(And in some cases a bare metal build may not be allowed to use the timer - eg secure-side images)

So not actually starting the system timer until it's referenced was a deliberate design choice. 

Maybe we could use a weak reference to initialise it only if it has been used by the app. Something like this should be possible on all toolchains:

     MBED_WEAK_REF OsTimer *init_os_timer(void);

     if (init_os_timer) { // ie "if init_os_timer is in the image because the app referenced it"
         init_os_timer();
     }

If that can be made to work portably, then that would seem ideal. (I think we have MBED_WEAK for a weak definition, but I've not seen us use weak references)

That was an initial "quick response". Going to study the actual report detail now.

kjbracey-arm, 2020-01-29:
I'm actually in progress of C++ chrono-ising this code, and actually, that chrono conversion _almost_ revealed the issue, but I hadn't quite grasped it.

Chrono `time_point` type checking would normally spot the incompatible absolute times. And it did! It effectively noted that the `set_wake_time` function was relying on `SysTimer::time_point` and `TickerClock::time_point` being the same. Chrono code had:

    template<class Period, bool IRQ>
    void SysTimer<Period, IRQ>::set_wake_time(time_point at)
    {
        //...
        duration ticks_to_sleep = at - get_tick();
        highres_time_point wake_time = at; // XXX

That only compiles because I marked `SysTimer::time_point` as being based on `SysTimer:: highres_time_point == TickerClock::time_point` - having the same epoch.

But that isn't actually true. I need to correct that to be different types, and then produce something that compiles for chrono. Once it does, we then have a correct algorithm and I can backport it.

I think solution 2 would be acceptable for bare metal, but I don't think it works for RTOS - it's more important for the OS tick count to equal the OsTimer tick count than the HAL timer, which means starting with "ticks = 0" on init, regardless of HAL timer value.

mprse, 2020-01-29:
> I think solution 2 would be acceptable for bare metal, but I don't think it works for RTOS - it's more important for the OS tick count to equal the OsTimer tick count than the HAL timer, which means starting with "ticks = 0" on init, regardless of HAL timer value.

I assumed that in the case of RTOS `OsTimer` is started at system startup and it is initializing the HAL ticker, so anyway os tick 0 and ticker 0 ms are consistent.

kjbracey-arm, 2020-01-29:
I don't think it's specified anywhere that the HAL timers necessarily start at zero. (Am I wrong?)

Plus there's nothing fundamentally stopping the bootloader or some other thing getting them going a few milliseconds before the RTOS init happens.

mprse, 2020-01-29:
You are right. There is no requirement which specifies that ticker starts counting from 0. Only that it continues counting from 0 after rollover:
https://github.com/ARMmbed/mbed-os/blob/f469f710bca16abd1c538e48a3bee9db0a8e47e8/hal/us_ticker_api.h#L83

kjbracey-arm, 2020-01-29:
So, for my chrono version, the fix seems to be to store the starting time as an extra member

    const highres_time_point _epoch;

Then the `set_wake_time` needs to do

    highres_time_point wake_time = _epoch + at.time_since_epoch();

And that now compiles, with the high-res and `SysTimer` clocks decoupled, so that should mean it's correct (right?) :)

I'll make a patch for the current version making the equivalent non-chrono change.

PR #12326 (2020-01-31): Correct SysTimer absolute time calculations

### Summary of changes 

`SysTimer::set_wake_time` incorrectly assumed that the `SysTimer`s tick count and the underlying HAL timer had the same zero base. This normally holds, at least approximately, in RTOS builds where the HAL timer starts from zero at the same time the SysTimer is initialised.

But in bare metal builds, the HAL timer could be started some time before the SysTimer, giving a significant discrepancy.

Beyond that, there's no requirement for HAL timers to start from zero in the spec.

Record the HAL timer start time to get the conve [... cut]
closed 2020-01-31 by kjbracey-arm