ARMmbed/mbed-os #12325: Cellular Contrext semaphore release twotimes

variablo87, 2020-01-29:
### Description of defect

If the I use `AT_CellularContext::connect()` with `_is_blocking=true` and the State Machine return (to cellular_callback) with a Error after all retrys (final_try), the `_semaphore` is released twotimes. 

```c++
void AT_CellularContext::cellular_callback(nsapi_event_t ev, intptr_t ptr)
{
[...]
if (data->final_try) {
     if (_current_op != OP_INVALID) {
        _semaphore.release();
     }
}
[...]
 if (_is_blocking) {
            if (_cb_data.error != NSAPI_ERROR_OK) {
                // operation failed, release semaphore
                if (_current_op != OP_INVALID) {
                    _current_op = OP_INVALID;
                    _semaphore.release();
                }
            } else {
```
At the second Connect after two releases, the function `AT_CellularContext::check_operation` will not wait for the State Machine!!!!

After this i get no correct `connect()` any more. 
Why the semaphore is release after final_try? 
The semaphore is used only at `_is_blocking` Mode. So it should released only at `_is_blocking` Mode.

My solution is:

```c++
void AT_CellularContext::cellular_callback(nsapi_event_t ev, intptr_t ptr)
{
[...]
//if (data->final_try) {
//     if (_current_op != OP_INVALID) {
//        _semaphore.release();
//     }
//}
[...]
 if (_is_blocking) {
            if ((data->final_try)||(_cb_data.error != NSAPI_ERROR_OK)) {
                // operation failed, release semaphore
                if (_current_op != OP_INVALID) {
                    _current_op = OP_INVALID;
                    _semaphore.release();
                }
            } else {
```

#### Target(s) affected by this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-99.99.99

#### How is this defect reproduced ? 
You can reproduces this this with https://github.com/ARMmbed/mbed-os-example-cellular
and you have no Cellular Network in your Office.

0xc0170, 2020-02-06:
cc @ARMmbed/mbed-os-wan

AriParkkila, 2020-02-17:
@variablo87 Thanks for reporting this. Your fix looks good to me. Can you make a pull request?

PR #12507 (2020-03-09): Cellular: release _semaphore only once

### Summary of changes 

Fixes #12325
The semaphore is used only at _is_blocking Mode. So it should released only at _is_blocking Mode.

### Documentation 
see issue #12325 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-03-09 by 0xc0170