ARMmbed/mbed-os #12337: BG96 cellular modem doesn't support IPV4V6 ?

jeromecoutant, 2020-01-30:
### Description of defect

From
https://github.com/ARMmbed/mbed-os/blob/44add981ff32f72eee6e99cc3b48d32011d06fec/features/cellular/framework/AT/AT_CellularDevice.h#L57
Setting PROPERTY_IPV4V6_PDP_TYPE means modem support IPV4 and IPV6 simultaneously

Ex with BG96:
https://github.com/ARMmbed/mbed-os/blob/44add981ff32f72eee6e99cc3b48d32011d06fec/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96.cpp#L59-L61
IPV4 OK, IPV6 OK, but not IPV4V6, so not simultaneously

Issue is while setting a new context:
https://github.com/ARMmbed/mbed-os/blob/44add981ff32f72eee6e99cc3b48d32011d06fec/features/cellular/framework/AT/AT_CellularContext.cpp#L446-L448

condition is wrong, we should not set IPV4V6 when both IPV4 and IPV6 are supported ?

#### Target(s) affected by this defect ?

BG96 cellular modem

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

jeromecoutant, 2020-02-06:
@AriParkkila

0xc0170, 2020-02-06:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2020-02-06:
Newer FW version of BG96 (BG96 Revision: BG96MAR02A07M1G) has the support for it

closed 2020-02-06 by AnttiKauppila

jeromecoutant, 2020-02-06:
This is not a question of BG96 support or not.
Question is:
- in 1 file, you can explicitly specify PROPERTY_IPV4V6_STACK to 0 meaning that you don't support both IPV4 or V6 as explained
- in other file, PROPERTY_IPV4V6_STACK is set to 1...