ARMmbed/mbed-os #12346: Unit Tests - Test #53: platform-ATCmdParser - .test_ATCmdParser_recv failing

ladislas, 2020-01-31:
### Description of defect

Test 53 failing on macOS Catalina with a fresh clone of master.

```console
test 53
      Start 53: platform-ATCmdParser

53: Test command: /Users/ladislas/dev/github/mbed-os/UNITTESTS/build/platform-ATCmdParser
53: Test timeout computed to be: 1500
53: Running main() from gmock_main.cc
53: [==========] Running 13 tests from 1 test case.
53: [----------] Global test environment set-up.
53: [----------] 13 tests from test_ATCmdParser
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_create
53: [       OK ] test_ATCmdParser.test_ATCmdParser_create (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_set_timeout
53: [       OK ] test_ATCmdParser.test_ATCmdParser_set_timeout (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_process_oob
53: [       OK ] test_ATCmdParser.test_ATCmdParser_process_oob (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_flush
53: [       OK ] test_ATCmdParser.test_ATCmdParser_flush (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_write
53: [       OK ] test_ATCmdParser.test_ATCmdParser_write (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_set_delimiter
53: [       OK ] test_ATCmdParser.test_ATCmdParser_set_delimiter (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_read
53: [       OK ] test_ATCmdParser.test_ATCmdParser_read (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_debug_on
53: [       OK ] test_ATCmdParser.test_ATCmdParser_debug_on (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_abort
53: [       OK ] test_ATCmdParser.test_ATCmdParser_abort (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_printf
53: [       OK ] test_ATCmdParser.test_ATCmdParser_printf (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_send
53: [       OK ] test_ATCmdParser.test_ATCmdParser_send (0 ms)
53: [ RUN      ] test_ATCmdParser.test_ATCmdParser_recv
53/54 Test #53: platform-ATCmdParser ......................................Bus error***Exception:   0.15 sec
test 54
      Start 54: platform-CircularBuffer

54: Test command: /Users/ladislas/dev/github/mbed-os/UNITTESTS/build/platform-CircularBuffer
54: Test timeout computed to be: 1500
54: Running main() from gmock_main.cc
54: [==========] Running 1 test from 1 test case.
54: [----------] Global test environment set-up.
54: [----------] 1 test from TestCircularBuffer
54: [ RUN      ] TestCircularBuffer.constructor
54: [       OK ] TestCircularBuffer.constructor (0 ms)
54: [----------] 1 test from TestCircularBuffer (0 ms total)
54:
54: [----------] Global test environment tear-down
54: [==========] 1 test from 1 test case ran. (1 ms total)
54: [  PASSED  ] 1 test.
54/54 Test #54: platform-CircularBuffer ...................................   Passed    0.14 sec

98% tests passed, 1 tests failed out of 54

Total Test time (real) =   7.76 sec

The following tests FAILED:
	 53 - platform-ATCmdParser (Bus error)
Errors while running CTest
```

I've narrowed it down to this line:

```cpp
EXPECT_TRUE(at1.recv("%c %d %x %s\r\n%c %d %x %s\r\n", &c, &intval, &hexval, &text));
```

#### Target(s) affected by this defect ?

unit tests

#### Toolchain(s) (name and version) displaying this defect ?

```console
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

#### What version of Mbed-os are you using (tag or sha) ?

latest master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```console
$ mbed --version
1.10.2
```

#### How is this defect reproduced ?

maciejbocianski, 2020-01-31:
@jamesbeyond

0xc0170, 2020-02-06:
Only on MacOS ? we haven't seen this failures, have we? cc @ARMmbed/mbed-os-wan

maciejbocianski, 2020-02-06:
@jamesbeyond ping

jamesbeyond, 2020-02-06:
Yep, I can reproduce this issue. the test got a segmentation fault error on Mac, but works fine with ubuntu. I'll try to figure out the root cause of this

jamesbeyond, 2020-02-07:
however, this just happened as well: https://github.com/ARMmbed/mbed-os/issues/12390

jamesbeyond, 2020-02-10:
The issue has been fixed

closed 2020-02-10 by jamesbeyond