ARMmbed/mbed-os #12347: LPC1768 ThisThread::sleep_for() timing issue

star297, 2020-01-31:
### Description of defect

LPC1768 ThisThread::sleep_for() timing is incorrect.
For instance ThisThread::sleep_for(500) results in sleep time of 5 seconds instead of 500ms.
wait_us() function works correctly.

#### Target(s) affected by this defect ?
LPC1768

#### Toolchain(s) (name and version) displaying this defect ?

Mbed online revision 7208

#### What version of Mbed-os are you using (tag or sha) ?

5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed online

#### How is this defect reproduced ? 

Run this snip to demonstrate the issue

Timer t1;
int getTime_delay;

int main(){ 
       
    t1.start();    
    while(1){            
        t1.reset(); 
        ThisThread::sleep_for(500);
        getTime_delay = t1.read_us();        
        pc.printf("Thread Wait Time: %d us \r\n",getTime_delay);
        
        t1.reset();
        wait_us(500000);
        getTime_delay = t1.read_us();        
        pc.printf("wait_us Time:     %d us \r\n\n",getTime_delay);
          
    }
}

0xc0170, 2020-02-07:
cc @ARMmbed/team-nxp @toyowata

toyowata, 2020-02-08:
@star297 Thanks you for your report.

I've quickly tested with your test case and got result below:

```
Thread Wait Time: 499403 us 
wait_us Time:     500008 us 

Thread Wait Time: 499361 us 
wait_us Time:     500012 us 

Thread Wait Time: 499320 us 
wait_us Time:     500009 us 

Thread Wait Time: 499382 us 
wait_us Time:     500012 us 

Thread Wait Time: 499340 us 
wait_us Time:     500009 us 

...
```

Both `ThisThread::sleep_for` and `wait_us` functions work as expected.

I thought the LPC1768 interface firmware had an issue about semihosting interface.  Can you check your firmware of the LPC1768?

https://os.mbed.com/platforms/mbed-LPC1768/#firmware

star297, 2020-02-08:
You are correct, it was the interface firmware.
I would not have thought to look there.
I'll close this issue.

closed 2020-02-08 by star297