ARMmbed/mbed-os #12348: K64F flash bank swapping reboots the board unexpectedly

leszek-grzegorek, 2020-01-31:
### Description of defect
The flash swap handling functionality included in K64F flash driver (fsl_flash.c) sometimes cause the sudden board reboot. The reproducibility is about 20%.
The issue seems to be timing related - when FLASH_Swap() function goes through swap machine states, the reboot takes place in Ready state on call to FLASH_SwapControl() to set next Update state (the call right after swap indicator address validation).
Putting a simple busy wait with wait_us() for e.g. 1s between both function calls described above solved the reboot issue.

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-7-2018-q2

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.0-rc3 (64853b354fa188bfe8dbd51e78771213c7ed37f7)
    
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.8.1

#### How is this defect reproduced ?
The issue can be reproduced by cyclic software flashing into unused (upper half) bank and triggering flash bank swap. The pseudo-code looks as follows :

```
#define FLASH_SIZE               0x100000
#define FLASH_SECTOR_SIZE        0x1000
#define SWAP_INDICATOR_ADDR      ((FLASH_SIZE >> 1) - FLASH_SECTOR_SIZE)

int main()
{
      flash_t flash;

      flash_init(&flash);

      initialize_network();

      get_software_bin_blocks_of_page_size_and_flash_it_into_upper_half();

      // make flash bank swap
      FLASH_Swap(&flash.flash_config, SWAP_INDICATOR_ADDR, kFLASH_SwapFunctionOptionEnable);

      printf("rebooting....\n");
      NVIC_SystemReset();
}
```

0xc0170, 2020-02-06:
This is 3rd party driver related (flash in this case) ? You might get quicker response on NXP forums (as it does not relate to Mbed OS. Although I can see flash_init call there).

cc @ARMmbed/team-nxp

maclobdell, 2020-02-06:
I don't think the Kinetis flash swap feature has been tested when running Mbed OS, but NXP forums might be able to help.  
By the way, a long time ago when I was at Freescale, I wrote an application note about this feature. 
http://cache.freescale.com/files/microcontrollers/doc/app_note/AN4533.pdf

leszek-grzegorek, 2020-02-07:
@0xc0170 : yes, in fact the flash swap functionality is provided with flash driver only and it is Freescale copyrighted.
@maclobdell : I was afraid that this part have not been tested in Mbed OS, because this issue would have been noticed earlier. The document you linked I have read before I realized that swap functionality is available in driver. BTW, I noticed is that FLASH_Swap() driver function does not realize the swap procedure exactly as it has been denoted in the document - in fact FLASH_Swap() goes through all the states with one shot, while it should stop on Update-Erased state and give the possibility to flash upper half and then be called second time to set the Complete state.

Can I count on any help regarding this use case in Mbed OS ?

mmahadevan108, 2020-02-07:
The only thing I could suggest is to try updating the SDK flash driver in mbed-os for K64F to the latest available at mcuxpresso.nxp.com

leszek-grzegorek, 2020-02-10:
@mmahadevan108 : Ok, I will give it a try. Thank you.

0xc0170, 2020-02-24:
> Can I count on any help regarding this use case in Mbed OS ?

I'll close this issue as not related to Mbed OS Flash functionality. Still the issue can be updated.
@leszek-grzegorek You can update this issue once you retest the latest driver. It could be helpful for someone.

closed 2020-02-24 by 0xc0170

leszek-grzegorek, 2020-02-28:
Just to update the status : with flash driver from MCUXpresso SDK v.2.7.0 (FRDM-K64F) ported to mbed 5.15.0-rc3, the issue still exists.