ARMmbed/mbed-os #12351: Question: How to reinitialise UART for mbed_debug?

amitchone, 2020-01-31:
### Description of defect
Hi there, 

I am using the `debug()` function to print a bunch of info/error messages when using the debug build profile. Under specific circumstances I instantiate a `UARTSerial` object in the follow way, where the TX/RX pins are the same as used for the `mbed_debug` UART:

    UARTSerial *_serial;
    _serial = new UARTSerial(TX, RX, BAUD);

My question is a simple one. Is there an explicit function call I can make to re-instantiate the UART used by `mbed_debug` after deleting my `_serial` object? 

Thanks,
Adam

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

5.15.0 - 64853b354fa188bfe8dbd51e78771213c7ed37f7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
ARM Compiler 6.13
Clangd 8.0.1
Mbed Studio 0.8.1

#### How is this defect reproduced ? 
N/A

amitchone, 2020-01-31:
Or, is there an object I can use instead of the `UARTSerial` object that I create? I'm using it with the `ATCmdParser`.

AGlass0fMilk, 2020-02-04:
Hi @amitchone, you may want to check out `mbed_override_console` and `mbed_file_handle` here:

https://github.com/ARMmbed/mbed-os/blob/586341541c9fc1dc05dca154858570e2e0107c97/platform/mbed_retarget.h#L115-L181

It looks like `mbed_debug` uses `printf` to print information out through `stdout`. Using the above functions you can access/retarget the serial instance used for `stdout`.

Let me know if you have any other questions.

0xc0170, 2020-02-04:
Thanks @AGlass0fMilk  for the help.

I'll close this as this is a question (use forum please).

closed 2020-02-04 by 0xc0170

amitchone, 2020-02-04:
Thanks @AGlass0fMilk, `mbed_file_handle` was exactly what I was looking for.