ARMmbed/mbed-os #12352: Greentea SYNC_FAIL using PDMC 2.2.1 Bootloader

lauleehong, 2020-02-03:
### Description of defect
Having a project with PDMC 2.2.1 bootloader defined in mbed_app.json would cause greentea to fail with SYNC_FAIL issue. My workaround to get greentea working was to remove the bootloader definition in mbed_app.json. 

Tagging @soleilplanet 

```
Build successes:
  * NUCLEO_F767ZI::GCC_ARM::MBED-BUILD
  * NUCLEO_F767ZI::GCC_ARM::SRC-DATASTRUCTCONVERSION-TESTS-DATASTRUCTCONVERSION-UNIT_TEST
[mbed-777] Exec "mbedgt --test-spec ./BUILD/tests/NUCLEO_F767ZI/GCC_ARM-TINY_DEBUG/test_spec.json -n src-datastructconversion-* -V" in "/home/yapziqi/Documents/REPOSITORIES/manuca-core"
[1579591825.70][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
[1579591825.72][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
mbedgt: greentea test automation tool ver. 1.7.4
mbedgt: test specification file './BUILD/tests/NUCLEO_F767ZI/GCC_ARM-TINY_DEBUG/test_spec.json' (specified with --test-spec option)
mbedgt: using './BUILD/tests/NUCLEO_F767ZI/GCC_ARM-TINY_DEBUG/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	| platform_name | platform_name_unique | serial_port  | mount_point                | target_id                |
	|---------------|----------------------|--------------|----------------------------|--------------------------|
	| NUCLEO_F767ZI | NUCLEO_F767ZI[0]     | /dev/ttyACM0 | /media/yapziqi/NODE_F767ZI | 08180221012469363E60F013 |
mbedgt: processing target 'NUCLEO_F767ZI' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
	| platform_name | platform_name_unique | serial_port         | mount_point                | target_id                |
	|---------------|----------------------|---------------------|----------------------------|--------------------------|
	| NUCLEO_F767ZI | NUCLEO_F767ZI[0]     | /dev/ttyACM0:115200 | /media/yapziqi/NODE_F767ZI | 08180221012469363E60F013 |
mbedgt: test case filter (specified with -n option)
	src-datastructconversion-*
	test filtered in 'src-datastructconversion-tests-datastructconversion-unit_test'
mbedgt: running 1 test for platform 'NUCLEO_F767ZI' and toolchain 'GCC_ARM'
	use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
	'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -m NUCLEO_F767ZI -p /dev/ttyACM0:115200 -f "BUILD/tests/NUCLEO_F767ZI/GCC_ARM-TINY_DEBUG/src/DatastructConversion/TESTS/DatastructConversion/unit_test/unit_test.bin" -d /media/yapziqi/NODE_F767ZI -c default -t 08180221012469363E60F013 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1579591826.31][HTST][INF] host test executor ver. 0.0.12
[1579591826.31][HTST][INF] copy image onto target...
[1579591826.31][COPY][INF] Waiting up to 60 sec for '08180221012469363E60F013' mount point (current is '/media/yapziqi/NODE_F767ZI')...
[1579591842.97][HTST][INF] starting host test process...
[1579591842.97][CONN][INF] starting connection process...
[1579591842.97][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1579591842.97][CONN][INF] initializing serial port listener...
[1579591842.97][PLGN][INF] Waiting up to 60 sec for '08180221012469363E60F013' serial port (current is '/dev/ttyACM0')...
[1579591842.97][HTST][INF] setting timeout to: 60 sec
[1579591842.98][SERI][INF] serial(port=/dev/ttyACM0, baudrate=115200, read_timeout=0.01, write_timeout=5)
[1579591842.98][SERI][INF] reset device using 'default' plugin...
[1579591843.09][SERI][INF] waiting 1.00 sec after reset
[1579591844.09][SERI][INF] wait for it...
[1579591844.10][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1579591844.10][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1579591844.10][CONN][INF] sending preamble 'b329c46a-8541-40ff-9c8d-9a7533bd3914'
[1579591844.10][SERI][TXD] {{__sync;b329c46a-8541-40ff-9c8d-9a7533bd3914}}
[1579591844.11][CONN][WRN] UnicodeDecodeError encountered!
[1579591844.12][CONN][RXD]  Mbed Bootloader
[1579591844.12][CONN][RXD] [BOOT] ARM: 00000000000000000000
[1579591844.12][CONN][RXD] [BOOT] OEM: 00000000000000000000
[1579591844.12][CONN][RXD] [BOOT] Layout: 0 8009568
[1579591844.12][CONN][RXD] [BOOT] Application's start address: 0x8010400
[1579591844.12][CONN][RXD] [BOOT] Application's jump address: 0x8021265
[1579591844.12][CONN][RXD] [BOOT] Application's stack address: 0x20080000
[1579591844.12][CONN][RXD] [BOOT] Forwarding to application...
[1579591849.11][CONN][INF] Reset the part and send in new preamble...
[1579591849.11][SERI][INF] reset device using 'default' plugin...
[1579591849.22][SERI][INF] waiting 1.00 sec after reset
[1579591850.22][SERI][INF] wait for it...
[1579591850.22][CONN][INF] resending new preamble 'ac06031c-0176-400a-890f-7cdc67d33fb0' after 5.00 sec
[1579591850.22][SERI][TXD] {{__sync;ac06031c-0176-400a-890f-7cdc67d33fb0}}
[1579591855.22][CONN][INF] Reset the part and send in new preamble...
[1579591855.22][SERI][INF] reset device using 'default' plugin...
[1579591855.33][SERI][INF] waiting 1.00 sec after reset
[1579591856.33][SERI][INF] wait for it...
[... 51 lines cut ...]
[mbed-777] ERROR: "mbedgt" returned error.
```

#### Target(s) affected by this defect ?
Tested on NUCLEO_F767ZI

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM Compiler 6 (gcc-arm-none-eabi-6-2017-q2-update)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 5.14.2 (cf4f12a)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2
mbed-greentea 1.7.4

#### How is this defect reproduced ? 
Mbed-os  >= 5.14.2
mbed_app.json with the following settings: 
`            "target.features_add"                       : ["BOOTLOADER"],
            "target.bootloader_img"                     : "bootloader/mbed-bootloader-NUCLEO_F767ZI.bin",
            "target.header_offset"                      : "0x10000",
            "target.app_offset"                         : "0x10400"`

where the bootloader used was provided in https://github.com/ARMmbed/pelion-ready-example/tree/master/bootloader

soleilplanet, 2020-02-03:
@MarceloSalazar  @chris-styles would you please help? Thanks!

MarceloSalazar, 2020-02-04:
@lauleehong can you please test on Windows and share the logs?
We'd like to understand whether this is related to https://github.com/ARMmbed/mbed-os/issues/4301

lauleehong, 2020-02-04:
The above greentea test was ran on Ubuntu 18.04 LTS

loverdeg-ep, 2020-02-11:
@trowbridgec

lauleehong, 2021-01-19:
Adding mbed-bootloader 4.7.0 to mbed 6.5.0 application would also cause SYNC_FAILED issue if we run greentea.

lauleehong, 2021-01-25:
I've fixed this issue. If your using a mbed-bootloader in your application program, you have to remove all serial printouts in the bootloader code. The sync_fail by greentea happens when your program bootloader prints and sends the bootloader printouts to the host machine during greentea test (which wasn't expected those characters from the target board).

closed 2021-01-25 by lauleehong