ARMmbed/mbed-os #12361: OKDO odin driver - ARM in nightly error hciCmdAlloc not defined

0xc0170, 2020-02-04:
### Description of defect

There are failures in nightly for OKDO_ODIN_W2 target. It affects only ARM driver (the binary in the tree). This is the error:

```
[DEBUG] Errors: Error: L6218E: Undefined symbol hciCmdAlloc (referred from ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(hci_vendor_cmd.o)).
[DEBUG] Errors: Error: L6218E: Undefined symbol hciCmdSend (referred from ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(hci_vendor_cmd.o)).
```

The latest nightly can be found here :http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/458/build-ARM/FAIL/OKDO_ODIN_W2/

The latest update for the target was 2 months ago, means something has changed somewhere else. See https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2

I can't spot what has changed recently causing this. The cordio sources have not changed for some time neither recent changes in BLE. Also this failure is only reported for ARM toolchain, so should be toolchain specific.

cc @ARMmbed/team-ublox @ARMmbed/mbed-os-pan 

#### Target(s) affected by this defect ?

OKDO_ODIN_W2

#### Toolchain(s) (name and version) displaying this defect ?

only ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

not applicable

#### How is this defect reproduced ? 

Build blinky example for this target and ARM, should be reproducible always

MarceloSalazar, 2020-02-04:
I'm able to reproduce this issue on the OKDO_ODIN_W2 - it happens with Mbed OS master (0b7c78be8a2e663a8d5ffb1843637af23476c116), not 5.15.0.
There must be something that landed recently and has caused this build problem, but I can't say what (yet).

However, the UBLOX_EVK_ODIN_W2 (own & maintained by Ublox) compiles just fine. Both are inherited from MODULE_UBLOX_ODIN_W2.

Is there a way to track what has changed?

0xc0170, 2020-02-05:
Thanks for checking, I realized this might be reviewed by different team . What is the team on github to be mentioned here?

I can see this has already a fix 👍

MarceloSalazar, 2020-02-05:
> What is the team on github to be mentioned here?
I'm working on this.
Note this platform support comes from an MTB reference design.
We should be enabling build checks on PRs to avoid this kind of issues happening again. Can you help with this?

0xc0170, 2020-02-05:
> We should be enabling build checks on PRs to avoid this kind of issues happening again. Can you help with this?

@ARMmbed/mbed-os-test Can you please review this issue? We had recently an issue with CI failure propagation so there could be a breakage hidden like this one.

0xc0170, 2020-02-06:
Fixed via #12370

closed 2020-02-06 by 0xc0170