ARMmbed/mbed-os #12367: EthernetInterface on Arch_max board doesn't update connection status

zhiyong80, 2020-02-05:
### Description of defect
EthernetInterface is supposed to mange and report connection status automatically, this seems to be the case on K64F board but not on ARCH_MAX board. When ethernet cable is unplugged after EthernetInterface initialization, calling get_connection_status() still returns NSAPI_STATUS_GLOBAL_UPNSAPI_STATUS_GLOBAL_UP/1.

When an Ethernet cable is unplugged, EthernetInterface is supposed to report connection status as 3(connecting), and this is the case on K64F, but on arch_max board, the returned value is still 1.

#### Target(s) affected by this defect ?
Arch_max

#### Toolchain(s) (name and version) displaying this defect ?
gcc_arm 6 2017-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.8.4
#### How is this defect reproduced ? 
1. Clone mbed http-example
2. Replace the code inside main function with the following lines:
```
    NetworkInterface* network = connect_to_default_network_interface();
    if (!network) {
        printf("Cannot connect to the network, see serial output\n");
    }
    while (1)
    {
            printf("%d: Network Status = %d\r\n", time(NULL), network->get_connection_status());
            ThisThread::sleep_for(1000);
    }

```
3. Compile and load image onto arch_max board.
4. Connect a ethernet cable to arch)max board, reset board by pressing on board reset button.
5. Unplug the ethernet cable after a while.
The console output will look like this:
```
527: Network Status = 1
528: Network Status = 1
529: Network Status = 1
530: Network Status = 1
// Ethernet cable unplugged here
531: Network Status = 1
532: Network Status = 1
533: Network Status = 1
534: Network Status = 1
535: Network Status = 1
536: Network Status = 1
536: Network Status = 1
```
If the same code is compiled and run on K64F board, the output will be something like the following:
```
7: Network Status = 1
8: Network Status = 1
9: Network Status = 1
10: Network Status = 1
11: Network Status = 1
12: Network Status = 1
// Ethernet cable unplugged here
13: Network Status = 3
14: Network Status = 3
15: Network Status = 3
16: Network Status = 3
17: Network Status = 3
18: Network Status = 3
```

eg-astrouka, 2020-02-06:
Generated by CubeMX code gives correct link status for the same board:
```
PHY_BSR register value = 0x786D

// Ethernet cable unplugged here

PHY_BSR register value = 0x7849
Ethernet connection changed
netif_is_link_up = 0
netif_is_up = 1
```
Running mbed code gives PHY_BSR register value always = 0xFFFF

0xc0170, 2020-02-06:
cc @toyowata @ARMmbed/mbed-os-ipcore

zhiyong80, 2020-02-06:
Can anyone try it on NUCLEO_F429ZI or other STM32F4 based boards? Arch_max reuses a lot of code that was designed for another ethernet phy(lan8472) on DP83848. There might be some subtle differences. For one, MDIO on DP83848 actually requires a pullup resistor to work but once that resistor is installed, Arch_max based boards will stop working.

zhiyong80, 2020-02-08:
@toyowata Thanks for testing on F429ZI. Mine won't come in until Monday. 

From what I can see, arch_max uses the same EMAC driver as other STM32F4. That driver was designed for a different ethernet phy. 

Is there anyway to patch up the possible difference between DP83848 and LAN8472?

zhiyong80, 2020-02-10:
@eg-astrouka found the problem is 
line 46 in /features/netsocket/emac-drivers/TARGET_STM/stm32xx_emac.cpp
```
#define ETH_ARCH_PHY_ADDRESS    (0x00)
```
Pin PHYAD0 of LAN8742 on F429ZI board is pulled down therefore uses 0x00 as phy address. Pin PHYAD0 of DP83848 is internalled pulled up and let float on Arch_max board, therefore uses 0x01 as phy address.

Once line 46 changed to the following:
```
#define ETH_ARCH_PHY_ADDRESS    (0x01)
```
The test code running on arch_max can also report link status change upon cable unplug and plug back in,

Since /features/netsocket/emac-drivers/TARGET_STM/stm32xx_emac.cpp seems to be shared by all STM32 based ethernet board, and different ethernet phy on different boards use different phy address, it probably makes better sense to move PHY_ADDRESS definition into folder of each individual board.

0xc0170, 2020-02-24:
Closing as the fix 12405 was merged

closed 2020-02-24 by 0xc0170