ARMmbed/mbed-os #12371: EFM32: pinmap test fails

amq, 2020-02-05:
### Description of defect

`tests-mbed_hal-pinmap` always fails on EFM32. The LED shows the mbed crash pattern after I run it. Other tests are fine.

If I comment out `gpio_pinmap`, then the test fails at `gpio_irq_pinmap`, also with a crash.

Related to https://github.com/ARMmbed/mbed-os/pull/9449 and https://github.com/ARMmbed/mbed-os/pull/10644

#### Target(s) affected by this defect ?

Verified on `EFM32GG_STK3700`, `EFM32PG12_STK3402` and `EFM32GG11_STK3701`, most probably affects all EFM32 targets.

#### Toolchain(s) (name and version) displaying this defect ?

`GNU Tools Arm Embedded 9 2019-q4-major`

#### What version of Mbed-os are you using (tag or sha) ?

`master` at [`f73a62afbf4052b4da8c5b862ffb4708a80c1b6e`](https://github.com/ARMmbed/mbed-os/commit/f73a62afbf4052b4da8c5b862ffb4708a80c1b6e)

Also verified with:

`mbed-os-5.13.4` at [`1bf6b20df9d3cd5f29f001ffc6f0d0fcbbb96118`](https://github.com/ARMmbed/mbed-os/commit/1bf6b20df9d3cd5f29f001ffc6f0d0fcbbb96118)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

`mbed-cli 1.10.2`

#### How is this defect reproduced ? 

```
mbed test --run -t GCC_ARM -m EFM32PG12_STK3402 -DMBED_EXTENDED_TESTS=1 -n tests-mbed_hal-pinmap -v
mbedgt: greentea test automation tool ver. 1.7.4
mbedgt: test specification file '.\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1580903499.40][mbedls.lstools_base]No valid file found to update JLink device details
mbedgt: detected 1 device
        | platform_name     | platform_name_unique | serial_port | mount_point | target_id                |
        |-------------------|----------------------|-------------|-------------|--------------------------|
        | EFM32PG12_STK3402 | EFM32PG12_STK3402[0] | COM5        | D:          | 204003F00001DCA9D5A8EF33 |
mbedgt: processing target 'EFM32PG12_STK3402' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        | platform_name     | platform_name_unique | serial_port | mount_point | target_id                |
        |-------------------|----------------------|-------------|-------------|--------------------------|
        | EFM32PG12_STK3402 | EFM32PG12_STK3402[0] | COM5:115200 | D:          | 204003F00001DCA9D5A8EF33 |
mbedgt: test case filter (specified with -n option)
        tests-mbed_hal-pinmap
        test filtered in 'tests-mbed_hal-pinmap'
mbedgt: running 1 test for platform 'EFM32PG12_STK3402' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m EFM32PG12_STK3402 -p COM5:115200 -f "BUILD/tests/EFM32PG12_STK3402/GCC_ARM/TESTS/mbed_hal/pinmap/pinmap.bin" -e "TESTS\host_tests" -d D: -c default -t 204003F00001DCA9D5A8EF33 -r default -C 4 -R 2 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1580903500.19][HTST][INF] host test executor ver. 0.0.12
[1580903500.19][HTST][INF] copy image onto target...
[1580903500.19][COPY][INF] Waiting up to 60 sec for '204003F00001DCA9D5A8EF33' mount point (current is 'D:')...
[1580903500.21][mbedls.lstools_base]No valid file found to update JLink device details
        1 file(s) copied.
[1580903506.78][mbedls.lstools_base]No valid file found to update JLink device details
[1580903507.33][mbedls.lstools_base]No valid file found to update JLink device details
[1580903507.88][mbedls.lstools_base]No valid file found to update JLink device details
[1580903508.43][mbedls.lstools_base]No valid file found to update JLink device details
[1580903508.99][mbedls.lstools_base]No valid file found to update JLink device details
[1580903509.50][HTST][INF] starting host test process...
[1580903510.29][CONN][INF] starting connection process...
[1580903510.29][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1580903510.29][CONN][INF] initializing serial port listener...
[1580903510.29][PLGN][INF] Waiting up to 60 sec for '204003F00001DCA9D5A8EF33' serial port (current is 'COM5')...
[... 56 lines cut ...]
mbedgt: test suite 'tests-mbed_hal-pinmap' ........................................................... FAIL in 13.03 sec
        test case: 'pinmap - validation' ............................................................. FAIL in 0.02 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.5712273915
mbedgt: test suite report:
| target                    | platform_name     | test suite            | result | elapsed_time (sec) | copy_method |
|---------------------------|-------------------|-----------------------|--------|--------------------|-------------|
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-pinmap | FAIL   | 13.03              | default     |
mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
| target                    | platform_name     | test suite            | test case           | passed | failed | result | elapsed_time (sec) |
|---------------------------|-------------------|-----------------------|---------------------|--------|--------|--------|--------------------|
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-pinmap | pinmap - validation | 0      | 1      | FAIL   | 0.02               |
mbedgt: test case results: 1 FAIL
mbedgt: completed in 13.36 sec
mbedgt: exited with code 1
[mbed] Working path "C:\Users\anybo\toolsense\mbed-os" (program)
[mbed] Exec "mbedgt --test-spec .\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json -n tests-mbed_hal-pinmap -V" in "C:\Users\anybo\toolsense\mbed-os"
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\Users\anybo\toolsense\mbed-os"
       Command: "mbedgt --test-spec .\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json -n tests-mbed_hal-pinmap -V"
---
```

amq, 2020-02-06:
The solution is to change
```
    NC = (unsigned int) 0xFFFFFFFFUL
```
to
```
    NC = (int) 0xFFFFFFFF
```
in `PinNames.h`. I'll create a PR shortly.

0xc0170, 2020-02-06:
Thanks @amq !

stevew817, 2020-02-24:
Was the mentioned PR ever created? :)

0xc0170, 2020-02-24:
I don't think I've seen one.

closed 2020-02-24 by 0xc0170