ARMmbed/mbed-os #12374: EFM32: RTC test fails

amq, 2020-02-05:
### Description of defect

`tests-mbed_hal-rtc` always fails on _some_ EFM32 targets. Other tests apart of https://github.com/ARMmbed/mbed-os/issues/12371 and https://github.com/ARMmbed/mbed-os/issues/12373 are fine.

It errors out at the `RTC - sleep` case with:
```
missing __exit_event_queue event from host test and no result from host test, timeout...
```

Related to https://github.com/ARMmbed/mbed-os/commit/5590f2b4954bdbfa77d54b7189518ff4f0b82e22 and https://github.com/ARMmbed/mbed-os/pull/7009

#### Target(s) affected by this defect ?

Verified on `EFM32PG12_STK3402` and `EFM32GG11_STK3701`.

`EFM32GG_STK3700` is not affected (because it's backed by `BURTC` instead of `RTCC`?)

#### Toolchain(s) (name and version) displaying this defect ?

`GNU Tools Arm Embedded 9 2019-q4-major`

#### What version of Mbed-os are you using (tag or sha) ?

`master` at [`f73a62afbf4052b4da8c5b862ffb4708a80c1b6e`](https://github.com/ARMmbed/mbed-os/commit/f73a62afbf4052b4da8c5b862ffb4708a80c1b6e)

Also verified with:

`mbed-os-5.13.4` at [`1bf6b20df9d3cd5f29f001ffc6f0d0fcbbb96118`](https://github.com/ARMmbed/mbed-os/commit/1bf6b20df9d3cd5f29f001ffc6f0d0fcbbb96118)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

`mbed-cli 1.10.2`

#### How is this defect reproduced ? 

```
mbed test --run -t GCC_ARM -m EFM32PG12_STK3402 -DMBED_EXTENDED_TESTS=1 -n tests-mbed_hal-rtc -v
mbedgt: greentea test automation tool ver. 1.7.4
mbedgt: test specification file '.\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1580922380.90][mbedls.lstools_base]No valid file found to update JLink device details
mbedgt: detected 1 device
        | platform_name     | platform_name_unique | serial_port | mount_point | target_id                |
        |-------------------|----------------------|-------------|-------------|--------------------------|
        | EFM32PG12_STK3402 | EFM32PG12_STK3402[0] | COM5        | D:          | 204003F00001DCA9D5A8EF33 |
mbedgt: processing target 'EFM32PG12_STK3402' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        | platform_name     | platform_name_unique | serial_port | mount_point | target_id                |
        |-------------------|----------------------|-------------|-------------|--------------------------|
        | EFM32PG12_STK3402 | EFM32PG12_STK3402[0] | COM5:115200 | D:          | 204003F00001DCA9D5A8EF33 |
mbedgt: test case filter (specified with -n option)
        tests-mbed_hal-rtc
        test filtered in 'tests-mbed_hal-rtc'
mbedgt: running 1 test for platform 'EFM32PG12_STK3402' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m EFM32PG12_STK3402 -p COM5:115200 -f "BUILD/tests/EFM32PG12_STK3402/GCC_ARM/TESTS/mbed_hal/rtc/rtc.bin" -e "TESTS\host_tests" -d D: -c default -t 204003F00001DCA9D5A8EF33 -r default -C 4 -R 2 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1580922381.75][HTST][INF] host test executor ver. 0.0.12
[1580922381.75][HTST][INF] copy image onto target...
[1580922381.75][COPY][INF] Waiting up to 60 sec for '204003F00001DCA9D5A8EF33' mount point (current is 'D:')...
[1580922381.77][mbedls.lstools_base]No valid file found to update JLink device details
        1 file(s) copied.
[1580922387.87][mbedls.lstools_base]No valid file found to update JLink device details
[1580922388.41][mbedls.lstools_base]No valid file found to update JLink device details
[1580922388.95][mbedls.lstools_base]No valid file found to update JLink device details
[1580922389.46][mbedls.lstools_base]No valid file found to update JLink device details
[1580922389.98][mbedls.lstools_base]No valid file found to update JLink device details
[... 75 lines cut ...]
|---------------------------|-------------------|--------------------|---------|--------------------|-------------|
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | TIMEOUT | 72.3               | default     |
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
| target                    | platform_name     | test suite         | test case        | passed | failed | result  | elapsed_time (sec) |
|---------------------------|-------------------|--------------------|------------------|--------|--------|---------|--------------------|
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - accuracy   | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - enabled    | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - glitch     | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - init       | 1      | 0      | OK      | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - persist    | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - range      | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - sleep      | 0      | 0      | ERROR   | 0.0                |
| EFM32PG12_STK3402-GCC_ARM | EFM32PG12_STK3402 | tests-mbed_hal-rtc | RTC - write/read | 0      | 0      | SKIPPED | 0.0                |
mbedgt: test case results: 6 SKIPPED / 1 OK / 1 ERROR
mbedgt: completed in 72.64 sec
mbedgt: exited with code 1
[mbed] Working path "C:\Users\amq\mbed-os" (program)
[mbed] Exec "mbedgt --test-spec .\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json -n tests-mbed_hal-rtc -V" in "C:\Users\amq\mbed-os"
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\Users\amq\mbed-os"
       Command: "mbedgt --test-spec .\BUILD\tests\EFM32PG12_STK3402\GCC_ARM\test_spec.json -n tests-mbed_hal-rtc -V"
---
```

0xc0170, 2020-02-06:
cc @ARMmbed/team-silabs

amq, 2020-02-18:
Findings so far:
- if `RTC - sleep` is moved to the end of the suite, then the `RTC - range` case succeeds
- if there is `wait_ms(1)` right after `rtc_write()` in `RTC - sleep`, then this case succeeds
- if there is `wait_ms(1)` right after `rtc_write()` in `RTC - range`, then this case succeeds without moving `RTC - sleep`

amq, 2020-02-19:
Seems like a context switch is needed.

Also just noticed: https://github.com/ARMmbed/mbed-os/issues/8906, could be related, see also the two fixing PRs.

amq, 2020-02-21:
@stevew817 could you have a look?

stevew817, 2020-02-24:
I'll see whether I can find something in the implementation that would correlate with your findings. Stay tuned.

stevew817, 2020-02-24:
wait_us maps to a busy-wait instead of a thread sleep, so that would explain why there could be a difference between the two.
Also seeing that rtc_write as part of the RTCC implementation is writing to the RTCC directly and to lptick_offset, which could potentially cause an attached lp_ticker event to fire too early or too late. I think the solution may be to add another tracking variable to keep track of the RTC offset, instead of messing with the hardware counter which potentially has a capture/compare already attached.

I need to experiment a bit, but the solution seems within reach.

PR #12509 (2020-02-27): EFM32: RTCC bugfix for #12374
RTC unit tests were failing on devices that use RTCC for both RTC and LP Ticker implementation.

### Summary of changes 

Now using an RTCC retention register to keep track of user timebase for RTC API. RTC and LP Ticker implementations use the same counter, but they shouldn't share timebases.

Closes #12374.

### Documentation 

None

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [X] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Fe [... cut]
closed 2020-02-27 by 0xc0170