ARMmbed/mbed-os #12376: Issue running blinky applcation on RT1050

mmahadevan108, 2020-02-05:
### Description of defect
The [blinky application](https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky//file/d323dd088ba2/main.cpp/)  on line 21 puts the platform to sleep in a while loop. 

On RT1050 this will trigger a deep sleep. This causes failures with drag-n-drop operations as DAPLink uses soft-reset to halt the target and is unable to halt the target.

#### Target(s) affected by this defect ?
RT1050

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Onlline compiler and mbed-cli

#### How is this defect reproduced ? 
Compile and program the blinky application to RT1050. Future drag-n-drop operations will fail

mmahadevan108, 2020-02-05:
cc @maclobdell

0xc0170, 2020-02-06:
@mmahadevan108 what deep sleep state is it entering? Is there any more "shallow" deep sleep that would not break daplink functionality?

mmahadevan108, 2020-02-06:
DAPLink is using soft-reset to HALT the target, hard-reset to HALT the target does not work on MXRT1050.
I am already using one of the shallower deep sleep modes to meet mbed-os latency requirements.

JanneKiiskila, 2020-02-28:
@mmahadevan108 @ARMmbed/team-nxp  - should we then disable sleep on this target completely? Would something like: https://github.com/JanneKiiskila/mbed-os/commit/029f701f937f26d865d5fce7c9f2ff6d4c38b5c5 do the trick?

**Stability beats performance any day**. If sleep is not stable, we should not enable it.

mmahadevan108, 2020-02-28:
I would request for a wait before the example enters the infinite loop. This would give an opportunity after reset to halt the target and program a new binary.

JanneKiiskila, 2020-03-02:
Hi @mmahadevan108 , 

Unfortunately that is not a good solution for the Resource as a Service (RaaS) system. We have have a number jobs that run on shared HWs.

1. They allocate the device.
1. They flash it
1. Run the test job (which ever it may be) and then 
1. Release the device until a next job is needed. 

If it's a PR job - the time for the next job can be several hours. No "magic fixed waits" will fix that problem.

1. Either we get the DAPLink properly fixed, so that it can gain control & flash the device reliably OR
1. We disable the sleeps, so that device remain reachable like they should.

Those are unfortunately the only options we have.

mmahadevan108, 2020-03-02:
The addition of the delay I am referring to is only related to this blinky application and not the mbed-os test applications.

The latest DAPLink firmware should fix the test instabilities seen.

mmahadevan108, 2020-03-02:
[k20dx_mimxrt1050_evk_hyper_if_crc_legacy_0x8000.zip](https://github.com/ARMmbed/mbed-os/files/4276800/k20dx_mimxrt1050_evk_hyper_if_crc_legacy_0x8000.zip)

Latest DAPLink

maclobdell, 2020-04-30:
@mmahadevan108  - can you confirm - does the latest DAPLINK with hard reset fix this issue, or is it still a problem. If so, why?

maclobdell, 2020-05-01:
I received more information about this issue that clarifies things. 
The DAPlink fix to issue a hard reset when drag-and-drop is initiated does not fix this problem, here is why:

1. When drag-and-drop is initiated, Mahesh's new daplink firmware does hard reset 
2. Then later in the code it gets to the part that connects to the board by issuing a soft-reset and halt.

If the cpu goes to deep sleep between 1 and 2, then it fails to connect.

For typical applications that use deep sleep this is not a problem, because the hard reset is able to wake it up and unless it tries to go right back asleep in a short amount of time, the debugger can connect and halt the core.

Perhaps a fix for this could be something like this. it would add a little clutter, but ensure all boards could connect.

```
#ifndef MBED_DEBUG
    /* wait some time to avoid going to sleep too quickly */
    wait_us(2000*1000);  
#endif
```

bluePain, 2020-06-07:
I'm trying to catch up with all you in different topics related to the same problem and if I didn't get it wrong, device goes into deep sleep mode if the app is built in the release mode, right?

Wish I knew that before flashing the device 😄 

(Most probably I missed it but) what is the workaround for *unbricking* the EVKB?

bluePain, 2020-06-07:
Got it working back again with the help of [this response](https://github.com/mbedmicro/pyOCD/issues/808#issuecomment-593636500).

Flashed device with OFF-ON-OFF-ON boot configuration, and then flashed it again with OFF-ON-ON-OFF configuration.