ARMmbed/mbed-os #12390: ATHandlers Unittest generate a segfault on Mac OS and Windows

jamesbeyond, 2020-02-07:
### Description of defect
While testing on Unittest framework, on Mac OS and Windows, on an issue been noticed:
After https://github.com/ARMmbed/mbed-os/pull/12305 merged
Unittest throw segfault on `TstATHandler.test_ATHandler_read_bytes`
@kivaisan 

The Segfault only happen on Mac OS and Windows, Ubuntu seems fine, that's why it didn't catch by CI. Maybe we need to run our CI unittest on Windows platform as well @VeliMattiLahtela 

#### Target(s) affected by this defect ?
Host (X86) Tests

#### Toolchain(s) (name and version) displaying this defect ?
gcc (x86_64-posix-seh-rev1, Built by MinGW-W64 project) 7.2.0

#### What version of Mbed-os are you using (tag or sha) ?

Latest Master
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.8.2

#### How is this defect reproduced ? 
```
mbed test --unittest
```

kivaisan, 2020-02-10:
@jamesbeyond Fix is now merged. Please verify.

jamesbeyond, 2020-02-10:
the issue has been fixed

closed 2020-02-10 by jamesbeyond