ARMmbed/mbed-os #12412: nsapi_ppp_connect failed on EC21

matsujirushi, 2020-02-12:
### Description of defect

When [this sample code](https://github.com/ARMmbed/mbed-os-example-cellular) is run on the EC21 module, the following error occurs.

`nsapi_ppp_connect failed`

An error occurs the first time, but the second time automatically retried succeeds.  
[log.txt](https://github.com/ARMmbed/mbed-os/files/4190593/log.txt)

#### Target(s) affected by this defect ?

QUECTEL/EC2X ([here](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/framework/targets/QUECTEL/EC2X))

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli with GCC_ARM

0xc0170, 2020-02-12:
cc @ARMmbed/mbed-os-wan

mikaleppanen, 2020-02-12:
@matsujirushi could you enable the PPP traces on /mbed-os/features/netsocket/ppp/mbed_lib.json and take a log file:

        "debug": {
            "help": "Enable debug traces for PPP",
            "value": false,                                                              <---------- change to true
            "macro_name": "PPP_DEBUG"
        },
        "enable-trace": {
            "help": "Enable PPP packet traces (traces every outbound and inbound packet)",
           "value": false                                                                 <--------- change to true
        },

Those will give more information why PPP fails.

matsujirushi, 2020-02-13:
@mikaleppanen I got logs and attached.
[log-good.txt](https://github.com/ARMmbed/mbed-os/files/4195925/log-good.txt)
[log-bad.txt](https://github.com/ARMmbed/mbed-os/files/4195927/log-bad.txt)

AriParkkila, 2020-02-14:
@matsujirushi It looks like modem is still looking for APN in failure cases. Can you set `"trace-level": "TRACE_LEVEL_DEBUG"` to see why it's goes to "Attaching network" state even though there is still "RSSI unknown". You could also clear excessive context (`AT+CGDCONT=CID`), especially the context with ID 2 as it seems to be active (+CGACT: 2,1) before the modem is attached to the network.

matsujirushi, 2020-02-17:
@AriParkkila attached.
[debug.txt](https://github.com/ARMmbed/mbed-os/files/4212198/debug.txt)

AriParkkila, 2020-02-17:
@matsujirushi Thanks for the debug logs. The modem reports PDP contexts incorrectly, and CellularStateMachine finds an active context as `+CGACT: 2,1` (in the log "[00013806ms][DBG ][CELL]: Found active context"). You need to clear extra contexts at least the context with ID 2 (see above). It may also help if you set `"cellular.clear-on-connect": true` in the mbed_app.json as that should clear extra contexts for you.

matsujirushi, 2020-02-18:
@AriParkkila Thanks for your advice!
It' fine after to set `"cellular.clear-on-connect": true` .
[fine.txt](https://github.com/ARMmbed/mbed-os/files/4216818/fine.txt)

closed 2020-02-18 by matsujirushi

CamelShoko, 2020-11-24:
thanks works will
-------
1606200996742ms][INFO][CELL]: CellularContext connect
[1606200996748ms][INFO][CELL]: Start connecting (timeout 1000 ms)
[1606200996748ms][INFO][CELL]: AT TX ( 2): AT
[1606200996748ms][INFO][CELL]: AT TX ( 1): <cr>
[1606200997751ms][INFO][CELL]: AT TX ( 2): AT
[1606200997751ms][INFO][CELL]: AT TX ( 1): <cr>
[1606200998756ms][INFO][CELL]: Modem power ON (timeout 1000 ms)
[1606200999356ms][WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT
[1606200999356ms][WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT
[1606200999356ms][WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D
[1606200999363ms][WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D
[1606200999369ms][INFO][CELL]: AT TX ( 7): AT+IPR=
[1606200999375ms][INFO][CELL]: AT TX ( 6): 230400
[1606200999381ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201000471ms][INFO][CELL]: AT TX ( 4): AT&W
[1606201000471ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201001472ms][INFO][CELL]: AT TX ( 4): ATE0
[1606201001472ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201002574ms][INFO][CELL]: AT TX ( 7): AT+IPR=
[1606201002574ms][INFO][CELL]: AT TX ( 6): 230400
[1606201002575ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201003676ms][INFO][CELL]: AT TX ( 4): AT&W
[1606201003676ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004678ms][INFO][CELL]: AT TX ( 4): ATE0
[1606201004678ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004900ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[1606201004900ms][INFO][CELL]: AT TX ( 9): AT+CMEE=1
[1606201004901ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004907ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[1606201004907ms][INFO][CELL]: AT TX ( 8): AT+CFUN?
[1606201004913ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004919ms][INFO][CELL]: AT RX (18): <cr><ln>+CFUN: 1<cr><ln><cr><ln>OK<cr><ln>
[1606201004925ms][INFO][CELL]: AT TX (17): AT+QSCLKEX=1,1,10
[1606201004931ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004937ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[1606201004943ms][INFO][CELL]: AT TX ( 7): AT+CGMI
[1606201004949ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004955ms][INFO][CELL]: AT RX (17): <cr><ln>Quectel<cr><ln><cr><ln>OK<cr><ln>
[1606201004967ms][INFO][CELL]: Modem manufacturer: Quectel
[1606201004973ms][INFO][CELL]: AT TX ( 7): AT+CGMM
[1606201004979ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004985ms][INFO][CELL]: AT RX (16): <cr><ln>EC200S<cr><ln><cr><ln>OK<cr><ln>
[1606201004991ms][INFO][CELL]: Modem model: EC200S
[1606201004997ms][INFO][CELL]: AT TX ( 7): AT+CGMR
[1606201004003ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004009ms][INFO][CELL]: AT RX (29): <cr><ln>EC200SCNAAR01A09M16<cr><ln><cr><ln>OK<cr><ln>
[1606201004015ms][INFO][CELL]: Modem revision: EC200SCNAAR01A09M16
[1606201004027ms][INFO][CELL]: Modem ready
[1606201004027ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[1606201004033ms][INFO][CELL]: AT TX ( 8): AT+CPIN?
[1606201004039ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201004045ms][INFO][CELL]: AT RX (18): <cr><ln>+CME ERROR: 14<cr><ln>
[1606201004057ms][WARN][CELL]: AT error code 14
[1606201004057ms][INFO][CELL]: SIM busy
[1606201005063ms][INFO][CELL]: Waiting for SIM (err while reading)...
[1606201005075ms][INFO][CELL]: Continue after 1 seconds
[1606201005081ms][INFO][CELL]: AT RX ( 7): <cr><ln>RDY<cr><ln>
[1606201006081ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[1606201006081ms][INFO][CELL]: AT TX ( 8): AT+CPIN?
[1606201006082ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201006088ms][INFO][CELL]: AT RX (18): <cr><ln>+CME ERROR: 14<cr><ln>
[1606201006088ms][WARN][CELL]: AT error code 14
[1606201006094ms][INFO][CELL]: SIM busy
[1606201006100ms][INFO][CELL]: Waiting for SIM (err while reading)...
[1606201006106ms][INFO][CELL]: Continue after 2 seconds
[1606201008112ms][INFO][CELL]: Setup SIM (timeout 1000 ms)
[1606201008112ms][INFO][CELL]: AT TX ( 8): AT+CPIN?
[1606201008113ms][INFO][CELL]: AT TX ( 1): <cr>
[... 92 lines cut ...]
[1606201022172ms][INFO][CAT1]: Cellular:ntp_sync:82: ntp time [Tue Nov 24 06:57:02 2020]
[1606201022173ms][INFO][CELL]: CellularContext disconnect()
[1606201022179ms][INFO][CELL]: AT TX ( 8): AT+CMEE?
[1606201022180ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201022226ms][INFO][CELL]: AT RX (10): <cr><ln>NO CARRI
[1606201022226ms][INFO][CELL]: AT RX ( 4): ER<cr><ln>
[1606201023180ms][WARN][CELL]: AT timeout
[1606201023180ms][WARN][CELL]: AT timeout
[1606201023180ms][INFO][CELL]: AT TX ( 8): AT+CMEE?
[1606201023181ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201023183ms][INFO][CELL]: AT RX (15): <cr><ln>+CMEE: 1<cr><ln><cr><ln>O
[1606201023187ms][INFO][CELL]: AT RX ( 3): K<cr><ln>
[1606201023193ms][INFO][CELL]: cellular_callback: PPP mode and NSAPI_STATUS_DISCONNECTED
[1606201023202ms][INFO][CELL]: AT TX ( 8): AT+CMEE?
[1606201023208ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201023214ms][INFO][CELL]: AT RX (18): <cr><ln>+CMEE: 1<cr><ln><cr><ln>OK<cr><ln>
[1606201023222ms][INFO][CELL]: CellularContext disconnected
[1606201023229ms][INFO][CELL]: cellular_callback: PPP mode and NSAPI_STATUS_DISCONNECTED
[1606201023238ms][INFO][CELL]: AT TX ( 8): AT+CMEE?
[1606201023244ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201023249ms][INFO][CELL]: AT RX (18): <cr><ln>+CMEE: 1<cr><ln><cr><ln>OK<cr><ln>
[1606201023258ms][INFO][CELL]: AT TX ( 9): AT+CFUN=0
[1606201023264ms][INFO][CELL]: AT TX ( 1): <cr>
[1606201023771ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[1606201023871ms][WARN][CAT1]: Cellular:main_task:117: shutdown elapsed [1698] ms