ARMmbed/mbed-os #12413: Nightly: baremetal example does not compile for predefined targets

0xc0170, 2020-02-12:
### Description of defect

There was PR 27 merged, here is it : https://github.com/ARMmbed/mbed-os-example-blinky-baremetal/pull/27. It was integrated yesterday. Nightly from the last night contains failures related to this changes.

```
#### End of STDOUT/STDERR ####

[EXAMPLES]> WARNING  Test for mbed-os-example-blinky will not be generated.
[EXAMPLES]> INFO     In folder 'mbed-os-example-blinky-baremetal'
[EXAMPLES]> INFO     Compiling mbed-os-example-blinky-baremetal LPC1768 ARM
[EXAMPLES]> INFO     Executing command 'mbed-cli compile -t ARM -m LPC1768 -j 0 --profile develop --profile mbed-os/tools/profiles/extensions/minimal-printf.json'...

#### STDOUT ####

Could not compile for LPC1768: small C library option not supported for this target.
[mbed] Working path "/builds/ws/mbed-os-ci_build-ARM-nightly/examples/mbed-os-example-blinky-baremetal" (program)

```

The latest nightly: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/464/

I can see LPC1768/ARCH PRO failing at least for ARM/IAR.

cc @ARMmbed/mbed-os-core

#### Target(s) affected by this defect ?

LPC1768/ARCH PRO

#### Toolchain(s) (name and version) displaying this defect ?

ARM, IAR

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Does not matter

#### How is this defect reproduced ? 

Build failing examples

evedon, 2020-02-12:
Indeed. Small C library is only supported in a few targets at the moment https://github.com/ARMmbed/mbed-os/blob/master/TESTS/configs/baremetal.json/#L41-L58
 "mbed_app.json` can be modified similarly as above until the feature is supported on all targets.

0xc0170, 2020-02-12:
@evedon Can you send PR fixing it ?

0xc0170, 2020-02-12:
K66F supports it ? @michalpasztamobica was not able to build it, see https://github.com/ARMmbed/mbed-os/issues/12414#issuecomment-585202621

evedon, 2020-02-12:
There is no support for small C library with IAR.

0xc0170, 2020-02-12:
True, so baremetal example should not be built with IAR then, this should be fixed in Mbed OS. I'll send PR shortly

0xc0170, 2020-02-12:
fix #12418

0xc0170, 2020-02-12:
Fixed via https://github.com/ARMmbed/mbed-os-example-blinky-baremetal/pull/28

closed 2020-02-12 by 0xc0170