ARMmbed/mbed-os #12414: Nightly: K66F and NUCLEO_F429ZI fail to build some examples

0xc0170, 2020-02-12:
### Description of defect

The logs show failures for K66F and NUCLEO_F429ZI in the latest nightly, see http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/464/build-IAR/FAIL/

However, I can't find any code change in the examples within the last day, neither in Mbed OS that could affect these. Previous nightly does not contain these so something had to change within the last 24 hours.

There are at least 2 different tables, search for `Passed Example Compilation` - one has all passing but baremetal, another one nanostack and mesh as well. I can't spot the difference. The logs are huge and finding two similar runs make it harder to find it.

cc @ARMmbed/mbed-os-test 

#### Target(s) affected by this defect ?

NUCLEO_F429ZI and K66F

#### Toolchain(s) (name and version) displaying this defect ?

IAR

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-02-12:
K66F , IAR (only IAR fails):

```
[Error] static_6lowpan_config.h@11,38: [Pe020]: identifier "MBED_CONF_APP_PSK_KEY" is undefined
[Error] static_6lowpan_config.h@12,27: [Pe020]: identifier "MBED_CONF_APP_TLS_PSK_KEY" is undefined
[Error] static_6lowpan_config.h@19,23: [Pe020]: identifier "MBED_CONF_APP_PSK_KEY_ID" is undefined
[Error] static_6lowpan_config.h@20,27: [Pe020]: identifier "MBED_CONF_APP_PAN_ID" is undefined
[Error] static_6lowpan_config.h@26,32: [Pe020]: identifier "MBED_CONF_APP_RF_CHANNEL" is undefined
[Error] static_6lowpan_config.h@27,32: [Pe020]: identifier "MBED_CONF_APP_RF_CHANNEL_PAGE" is undefined
[Error] static_6lowpan_config.h@28,32: [Pe020]: identifier "MBED_CONF_APP_RF_CHANNEL_MASK" is undefined
[Error] static_6lowpan_config.h@29,31: [Pe020]: identifier "MBED_CONF_APP_RPL_INSTANCE_ID" is undefined
[Error] static_6lowpan_config.h@30,22: [Pe020]: identifier "MBED_CONF_APP_RPL_IDOUBLINGS" is undefined
[Error] static_6lowpan_config.h@31,33: [Pe020]: identifier "MBED_CONF_APP_RPL_K" is undefined
[Error] static_6lowpan_config.h@32,37: [Pe020]: identifier "MBED_CONF_APP_RPL_MAX_RANK_INC" is undefined
[Error] static_6lowpan_config.h@33,25: [Pe020]: identifier "MBED_CONF_APP_RPL_MIN_HOP_RANK_INC" is undefined
[Error] static_6lowpan_config.h@34,37: [Pe020]: identifier "MBED_CONF_APP_RPL_IMIN" is undefined
[Error] static_6lowpan_config.h@35,34: [Pe020]: identifier "MBED_CONF_APP_RPL_DEFAULT_LIFETIME" is undefined
[Error] static_6lowpan_config.h@36,24: [Pe020]: identifier "MBED_CONF_APP_RPL_LIFETIME_UNIT" is undefined
[Error] static_6lowpan_config.h@37,24: [Pe020]: identifier "MBED_CONF_APP_RPL_PCS" is undefined
[Error] static_6lowpan_config.h@38,35: [Pe020]: identifier "MBED_CONF_APP_RPL_OCP" is undefined
[Error] static_6lowpan_config.h@39,35: [Pe020]: identifier "MBED_CONF_APP_RA_ROUTER_LIFETIME" is undefined
[Error] static_6lowpan_config.h@40,31: [Pe020]: identifier "MBED_CONF_APP_BEACON_PROTOCOL_ID" is undefined
[Error] static_6lowpan_config.h@42,43: [Pe020]: identifier "MBED_CONF_APP_TLS_PSK_KEY_ID" is undefined
[Error] static_6lowpan_config.h@43,34: [Pe020]: identifier "MBED_CONF_APP_BACKHAUL_DYNAMIC_BOOTSTRAP" is undefined
[Error] static_6lowpan_config.h@44,37: [Pe020]: identifier "MBED_CONF_APP_SHORT_MAC_ADDRESS" is undefined
[Error] static_6lowpan_config.h@46,26: [Pe020]: identifier "MBED_CONF_APP_PREFIX_FROM_BACKHAUL" is undefined
[Error] static_6lowpan_config.h@50,0: [Pe028]: expression must have a constant value
```

```
[Error] mesh_led_control_example.cpp@39,24: [Pe020]: identifier "MBED_CONF_APP_LED" is undefined
[Error] mesh_led_control_example.cpp@40,20: [Pe020]: identifier "MBED_CONF_APP_BUTTON" is undefined
[Error] mesh_led_control_example.cpp@41,25: [Pe020]: identifier "MBED_CONF_APP_RELAY_CONTROL" is undefined
[Error] mesh_led_control_example.cpp@210,0: [Pe020]: identifier "MBED_CONF_APP_BUTTON_MODE" is undefined
```

Same failures for NUCLEO_F429ZI. These are the failures there. Config failing but if we haven't changed the config in the examples, was this changed in the CI?

0xc0170, 2020-02-12:
cc @mbed-os-wan - just in case if they know what could have changed yesterday

0xc0170, 2020-02-12:
This is not just nightly, affects now CI PRs as well, see https://github.com/ARMmbed/mbed-os/pull/11942

0xc0170, 2020-02-12:
Note to all of you, we won't run any CI until this is fixed as build fails

michalpasztamobica, 2020-02-12:
I tried to build this locally with:
```
mbed compile -m K66F -t IAR -j 0 --profile develop --profile mbed-os/tools/profiles/extensions/minimal-printf.json -v
```
But every time the compilation succeeded. I tried example@5.15 and example@master each time with mbed-os at latest master.
CI must be doing something differently.

michalpasztamobica, 2020-02-12:
I can reproduce the other CI failure:
```
~/arm/mbed-os-example-blinky-baremetal (master)
$ mbed compile -m K66F -t IAR -j 0 --profile develop --profile mbed-os/tools/profiles/extensions/minimal-printf.json -v

Could not compile for K66F: small C library option not supported for this target.
```
Could this influence subsequent builds in any way?

0xc0170, 2020-02-12:
@michalpasztamobica baremetal failures are being tracked here #12413 . They should not affect but can't rule that one out.

0xc0170, 2020-02-12:
Most likely fix for #12413 will also resolve this. I'll close this one once confirmed.

closed 2020-02-12 by 0xc0170