ARMmbed/mbed-os #12416: EMACInterface::connect() always succeeds with an incorrect PHY address

timovanasten, 2020-02-12:
### Description of defect

We use a custom board based on the NUCLEOF767ZI, but we've used another PHY transceiver than the one on the NUCLEO for the Ethernet connectivity. We had to change the PHY address for this but noticed that `EMACInterface::connect()` always returns 0 (i.e. success) when using an incorrect PHY address.  I think the desired response should be an error in this case. We suspect that this is caused by the fact that when PHY address is configured incorrectly, the `HAL_ETH_ReadPHYRegister(ETH_HandleTypeDef *heth, uint16_t PHYReg, uint32_t *RegValue)` in stm32f7xx_hal_eth.c seems to always return `0b1111111111111111`.

#### Target(s) affected by this defect ?
NUCLEOF7xx, Custom targets based on STM32F7xx targets

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
For the NUCLEOF767ZI, change the ETH_ARCH_PHY_ADDRESS defined stm32xx_emac.cpp from `(0x00)` to `(0x01)`. Then call `NetworkInterface::get_default_instance().connect()`. This will always return 0.

0xc0170, 2020-02-12:
cc @ARMmbed/mbed-os-ipcore

michalpasztamobica, 2020-02-12:
Hi @timovanasten .
Please check https://github.com/ARMmbed/mbed-os/pull/12405. Seems like we may need to add another "target_override" for NUCLEOF767ZI?

timovanasten, 2020-02-12:
@michalpasztamobica That would definitely be a nice option to have. Right now we need to dig into the drivers ourselves which seems a bit hacky. It doesn't solve the issue about the return value of the connect() function not representing the actual situation if it is wrongly configured though.

michalpasztamobica, 2020-02-12:
You're right. But to check the return values of ST driver functions, we may need to ask someone from ST. @LMESTM , can you give us a hand with this? @0xc0170 , I can see `team-st-mcd` tag, but I am not sure what MCD stands for, so I don't dare to use it, so that I don't bother some innocent people...

jeromecoutant, 2020-02-12:
Hi
I also think #12405 is solving this issue ?
You then just need to add in your mbed_app json file something like
    "stm32-emac.eth-phyaddr": 1

jeromecoutant, 2020-02-12:
@michalpasztamobica 
Yes you can use @ARMmbed/team-st-mcd alias :-)
MCD is the ST entity in charge of STM32 support

michalpasztamobica, 2020-02-12:
@jeromecoutant , I think @timovanasten still has a point, that in case of invalid PHY_ADDR the value returned from `HAL_ETH_ReadPHYRegister` should be some error code, rather than `b'111111`...
The function returns `HAL_StatusTypeDef`, perhaps it would be good to detect the issue and return `HAL_ERROR    = 0x01U`?

jeromecoutant, 2020-02-12:
Something like this ?
https://github.com/jeromecoutant/mbed/commit/f6059c17473e37f76ae2893f450fbc43899113c0

timovanasten, 2020-02-12:
@jeromecoutant Just quickly checked out your commit, but it seems like the connect() method still returns 0 while using the wrong PHY address

timovanasten, 2020-02-12:
Just checked that `HAL_ETH_ReadPHYRegister` returns `HAL_OK` with a wrong PHY address

jeromecoutant, 2020-02-13:
@timovanasten 
Could you try https://github.com/jeromecoutant/mbed/commit/8abf2a76332966727e69281ab922384d5ffd02fe ?
Thx

timovanasten, 2020-02-16:
> @timovanasten
> Could you try [jeromecoutant@8abf2a7](https://github.com/jeromecoutant/mbed/commit/8abf2a76332966727e69281ab922384d5ffd02fe) ?
> Thx

I should be able to to that on Monday. I'll let you know.

timovanasten, 2020-02-17:
@jeromecoutant With wrong PHY address, I get:
```
[INFO][STETH1]: power_up
[INFO][STETH1]: low_level_init_successful
[ERR ][NWRK]: Initializing Ethernet interface failed with error code -3004, Retrying in 5000 milliseconds...
[ERR ][NWRK]: Initializing Ethernet interface failed with error code -3020, Retrying in 5000 milliseconds...
.
.
.
```
Where the last two lines are printed by my software, and the last line repeats.

With correct PHY address, I get
```
[INFO][STETH1]: power_up
[INFO][STETH1]: low_level_init_successful
[INFO][STETH1]: emac_link_state_cb set to true
[

++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x801E1BF
Error Value: 0x1
Current Thread: stm32_emac_thread  Id: 0x20005CAC Entry: 0x80134D9 StackSize: 0x200 StackMem: 0x200139F0 SP: 0x2007FF28
For more info, visit: https://mbed.com/s/error?error=0x80020125&tgt=NUCLEO_F767ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=2) reached maximum, system will halt after rebooting 
```

jeromecoutant, 2020-02-17:
So:
- in default mode, implementation is as expected: no connection with wrong PHY address, and working connection with correct PHY address
- in debug mode, implementation is OK with correct PHY address, but crashes with wrong PHY address

jeromecoutant, 2020-02-17:
@timovanasten 
Did you optimize stack size in your application ?
(maybe some sizes need to be increased when debug is on?)

timovanasten, 2020-02-17:
> So:
> 
> * in default mode, implementation is as expected: no connection with wrong PHY address, and working connection with correct PHY address
> * in debug mode, implementation is OK with correct PHY address, but crashes with wrong PHY address

@jeromecoutant That's correct. It seems to work fine when I disable debug mode. I didn't do anything with stack size optimization yet. Anything you would like me to try?

timovanasten, 2020-02-17:
> So:
> 
> * in default mode, implementation is as expected: no connection with wrong PHY address, and working connection with correct PHY address
> * in debug mode, implementation is OK with correct PHY address, but crashes with wrong PHY address

@jeromecoutant Not really. With mbed trace enabled on debug level, the application crashes on a correct PHY address. When I then disable the debug statements of mbed trace, the application connects as expected. 

With an incorrect PHY, the application doesn't connect with -3004 (as expected I would say). That was with mbed trace on debug and I expect the same behaviour when disabling debug, although I did not try that.

 I didn't do anything with stack size optimization yet. Anything you would like me to try?

jeromecoutant, 2020-02-18:
> Current Thread: stm32_emac_thread  Id: 0x20005CAC Entry: 0x80134D9 StackSize: 0x200 

This StackSize is configured here:
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_STM/TARGET_STM32F7/stm32xx_emac_config.h

Maybe you could increase the value and see ?

jeromecoutant, 2020-02-18:
You can also use latest branch: https://github.com/jeromecoutant/mbed/tree/PR_ETHERNET

timovanasten, 2020-02-20:
@jeromecoutant I've used your latest branch and increased the stack size to 1024. It seems to work as expected now!

**With correct PHY address**
```
[INFO][MSGR]: Starting connection to EMS...
[INFO][STE1]: power_up
[INFO][STE1]: PHY Addr 0 AutoNegotiation 1
[DBG ][STE1]: MAC Addr 00:80:e1:30:00:3e
[INFO][STE1]: ETH buffers : 4 Rx 4 Tx
[INFO][STE1]: low_level_init_successful
[INFO][STE1]: emac_link_state_cb set to true
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 86
[DBG ][STE1]: low_level_input len 62
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 130
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 546
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 217
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 546
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input len 130
[DBG ][STE1]: low_level_input len 217
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[INFO][MSGR]: Connection to EMS successful. Ready to receive messages
```

**With incorrect PHY address**
```
[INFO][MSGR]: Starting connection to EMS...
[INFO][STE1]: power_up
[INFO][STE1]: PHY Addr 1 AutoNegotiation 1
[DBG ][STE1]: MAC Addr 00:80:e1:30:00:3e
[INFO][STE1]: ETH buffers : 4 Rx 4 Tx
[INFO][STE1]: low_level_init_successful
[DBG ][STE1]: low_level_input len 70
[DBG ][STE1]: low_level_input len 217
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 546
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 70
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 546
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 86
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 546
[DBG ][STE1]: low_level_input len 566
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 60
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 86
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 87
[DBG ][STE1]: low_level_input no frame
[DBG ][STE1]: low_level_input len 168
[DBG ][STE1]: low_level_input no frame
[ERR ][NWRK]: Initializing Ethernet interface failed with error code -3004, Retrying in 5000 milliseconds...
```

jeromecoutant, 2020-02-20:
> ... and increased the stack size to 1024. 

Which profile do you use for compilation?
- default one: develop
- debug
?

timovanasten, 2020-02-20:
@jeromecoutant debug in this case

jeromecoutant, 2020-02-20:
OK!
So issue seems to occur with GCC + debug mode
(no issue with ARMC6 and debug...)

let's merge #12464, then I will add some workaround for that !

Thx

jeromecoutant, 2020-03-03:
> So issue seems to occur with GCC + debug mode

Hi

Wrong assumption...
Issue seems not linked to debug mode, but only with GCC and mbed trace feature.

See #12561

closed 2020-03-03 by 0xc0170

jeromecoutant, 2020-03-19:
@timovanasten 
I am going to push https://github.com/jeromecoutant/mbed/commit/488c9db81e891dd8f640b98b14ecca251c28c5d8
Is it still OK with you ?

timovanasten, 2020-03-19:
@jeromecoutant Yes totally fine! Go ahead

PR #12464 (2020-03-03): STM32 EMAC : add configuration choice and connection check
### Summary of changes 

Request/question started in https://forums.mbed.com/t/stm32f767-ethernet-with-dp83640/7149

This also fixes #12416 

So few corrections are proposed:
- Add some flexibility and be able to configure locally parameters value depending on chosen PHY
- Add several check before setting OK init and connection procedures

#### Impact of changes 

#### Migration actions required 

STM32 EMAC is more configurable for each application.
Default setting have been set for the supported ST boards.

### Documentation 

------ [... cut]