ARMmbed/mbed-os #12439: FlashIAP::init - Possible memory leak/OOM error

dustin-crossman, 2020-02-14:
### Description of defect
If a targets flash_init() returns an errror, FlashIAP::init() will catch this and set its own return code properly, however, it does not return early. As a result it will call get_page_size() and then use new to allocate a page.
This can lead to two problems:
1. If new allocates successfully a memory leak may occur because the user will not expect (and should not be expected) to call deinit() after a failed init.
2. If get_page_size() returns wrong/junk data new may be called with a very large value and cause an Out of Memory error.

Issue #2 was seen in practice while we were testing changes to our flash driver which caused it to fail on init and which therefore caused get_page_size() to return an uninitialized value.

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
master
sha: 3d038e55ee

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Modify a targets flash_init() to return failure.

0xc0170, 2020-02-17:
Thanks for the report, could you propose a fix? 

If we fail in init, we should not allocate anything and return an error. It would fix 1. and also 2. (we don't have there yet defined behavior but should be call deinit only on initialized flash otherwise undefined behavior - something similar to what serial already has defined: `Calling any function other than ::serial_init on am uninitialized or freed `serial_t`.`).

dustin-crossman, 2020-02-20:
I can get a PR in for a fix. Probably won't be until next week though.

closed 2020-03-05 by 0xc0170