ARMmbed/mbed-os #12452: u-blox ODIN-W2 error on cbMAIN_startOs

avinabmalla, 2020-02-17:
### Description of defect
I am trying to use Bluetooth SPP using the C API.
Calling cbMAIN_startOS after cbMAIN_initBt throws the following error in the error handler, and restarts the device.

Error Code: -1
File: cb_uart.c
Line: 2150

I also tried calling cbMAIN_initBt without cbMAIN_startOS . The callback in cbMAIN_initBt is never called.

I have also tried setting BLE_STACK_UBX in the targets.json file to enable u-blox stack rather than Cordio.

#### Target(s) affected by this defect ?
UBLOX_EVK_ODIN_W2
MTB_UBLOX_ODIN_W2

#### Toolchain(s) (name and version) displaying this defect ?
GNU Tools ARM Embedded 6 2017-q2-update

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ?
I tried initializing Bluetooth with the following code:
 
cbMAIN_initOS();
cbMAIN_registerErrorHandler(onError);
cbBT_UTILS_setInvalidBdAddr(&initParams.address);
initParams.leRole = cbBM_LE_ROLE_DISABLED;
initParams.maxOutputPower = cbBM_MAX_OUTPUT_POWER;
initParams.maxLinkKeysClassic = 25;
initParams.maxLinkKeysLe = 25;
cbMAIN_initBt(&initParams, initBt);
cbMAIN_driverLock();
cbMAIN_startOS();
cbMAIN_driverUnlock();

0xc0170, 2020-02-19:
cc @ARMmbed/team-ublox

aqib-ublox, 2020-02-19:
@avinabmalla we dont support ubx stack, only cordio stack is supported

0xc0170, 2020-02-24:
I close this as not supported (not technical issue here).

closed 2020-02-24 by 0xc0170