ARMmbed/mbed-os #12455: nightly/PRs: IAR build errors with an internal error

0xc0170, 2020-02-18:
### Description of defect

This currently affects PRs and it is also in the latest nightly (was not in the day before). Something has changed in CI or example?

This is the example failing `mbed-os-example-socket-stats` - cc @ARMmbed/mbed-os-core 

```
[ERROR] Internal error: [Front end]: assertion failed at: "/home/jenkins_ci01/slave_top/workspace/ArmReleaseBuild/core/translator/compiler_core/src/parser/edg/il.c", line 23626

Fatal error detected, aborting.

[mbed] Working path "/builds/ws/mbed-os-ci_build-IAR@3/examples/mbed-os-example-socket-stats" (program)
```

As nightly and also PRs (one is using master and one release branch), this does not indicate it is in the example itself but rather a change in CI within the last 15 hours or so?

cc @ARMmbed/mbed-os-test 

#### Target(s) affected by this defect ?

K64F

#### Toolchain(s) (name and version) displaying this defect ?

IAR

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

CI related

#### How is this defect reproduced ? 

I am testing locally , will update

0xc0170, 2020-02-18:
I am able to reproduce it on example master

```
Scan: mbed-os-example-socket-stats
Compile [  0.1%]: mbed_tz_context.c
Compile [  0.2%]: MCR20Drv.c
Compile [  0.3%]: at24mac.cpp
Compile [  0.4%]: main.cpp
[ERROR] Internal error: [Front end]: assertion failed at: "C:\j\workspace\Arm832Build\core\translator\compiler_core\src\parser\edg\il.c", line 23630

```

If I revert the last commit: https://github.com/ARMmbed/mbed-os-example-socket-stats/commit/22cdd957a8326a5e7de064a861602f323a068a68 , it builds

@evedon Something triggers IAR internal fault in the latest commit there.

0xc0170, 2020-02-18:
I found the offending line:

`mbed_stats_socket_t stats[MBED_CONF_NSAPI_SOCKET_STATS_MAX_COUNT] = {0};` If I remove `{0}`, no internal fault. 

cc @kjbracey-arm

0xc0170, 2020-02-18:
Found a fix, not ideal but will do the same functionality just avoiding the IAR error

0xc0170, 2020-02-18:
https://github.com/ARMmbed/mbed-os-example-socket-stats/pull/47 should fix it for now

kjbracey-arm, 2020-02-18:
I think the revised code is better - it's a big enough object to be worth not putting on the stack.

closed 2020-02-18 by evedon

0xc0170, 2020-02-18:
I also reported this to IAR via website support.