ARMmbed/mbed-os #12465: [PSOC6] AnalogIn test for checking reference voltage

heejungPark, 2020-02-19:
### Description of defect
We got a abnormal return value from read() or read_u16() in AnalogIn class.
I knew default reference voltage of this target is 3.3V, but it may be not true.
For example, when A0 pin is 0.58V, read() function outputs 0.242.
It means that reference voltage is about 2.4V....
Please confirm this issue and inform me.

#### Target(s) affected by this defect ?
CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
mbed-cli
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
Mbed-OS 5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2
GCC_ARM 6.3.1

#### How is this defect reproduced ? 
You can just run this example.
https://os.mbed.com/docs/mbed-os/v5.15/apis/analogin.html

0xc0170, 2020-02-20:
cc @ARMmbed/team-cypress

kyle-cypress, 2020-02-20:
The PSoC 6 ADC is currently configured to use the internal bandgap (1.2V) as its reference, which gives it a range of 0 - 2.4 V. To better align with the expectations of the mbed HAL, we intend to update this in a future release to use VDDA as the reference instead.

heejungPark, 2020-02-24:
Okay, thanks for your reply.

closed 2020-02-24 by heejungPark