ARMmbed/mbed-os #12474: Stack overflow occurs on EC21

matsujirushi, 2020-02-20:
### Description of defect

Sometimes stackoverflow occurs.

```
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x8021B63
Error Value: 0x1
Current Thread: ppp Id: 0x20009518 Entry: 0x81016000 StackSize: 0x80341C7 StackMem: 0xFFFFFFFF SP: 0x2001FF3C 
```

[good.txt](https://github.com/ARMmbed/mbed-os/files/4229502/good.txt)

[fail.txt](https://github.com/ARMmbed/mbed-os/files/4229503/fail.txt)

**Code snippet:**
```cpp
		// ネットワーク接続
		print_function("connect()\n");
		if (ctx->connect() != NSAPI_ERROR_OK) abort();

		// 接続基地局取得
		print_function("get_registration_params()\n");
		auto net = ctx->get_device()->open_network();
		CellularNetwork::registration_params_t param;
		if (net->get_registration_params(param) != NSAPI_ERROR_OK) abort();
		ctx->get_device()->close_network();
		auto lac = format("%x", param._lac);
		auto cid = format("%x", param._cell_id);
```

#### Target(s) affected by this defect ?

QUECTEL/EC2X ([here](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/framework/targets/QUECTEL/EC2X))

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli with GCC_ARM

0xc0170, 2020-02-24:
@matsujirushi What is the app setup - stack sizes per Thread, how many  threads, etc? What is the size of the stack for a thread that errors? Is the size sufficient for its needs?

Have you tried to enable stats (see https://os.mbed.com/blog/entry/Tracking-memory-usage-with-Mbed-OS/) to track the usage of memory?

AnttiKauppila, 2020-02-24:
What is the target board and how much it has memory?

matsujirushi, 2020-02-25:
@0xc0170 
I will try to enable stats. Wait a minute.

@AnttiKauppila 
I use STM32F405RG. Flash 1Mbytes, SRAM 192Kbytes.

AnttiKauppila, 2020-02-25:
@matsujirushi That should be plenty. 
It seems that your ppp thread's stack memory is too small
You should overwrite ppp.thread-stacksize (default: 816) to bigger in your mbed_app.json
@mikaleppanen Has something changed in netsocket/PPP? It seems that default stack size for PPP might not be enough currently?

matsujirushi, 2020-02-26:
@0xc0170 I got stats and find code.
[teraterm.log](https://github.com/ARMmbed/mbed-os/files/4253428/teraterm.log)

```
# Fine

[00013759ms]
Thread: 0x20001E24, Stack size: 1576 / 4096
Thread: 0x20008D40, Stack size: 1272 / 2048
Thread: 0x20001F74, Stack size: 88 / 512
Thread: 0x20001F30, Stack size: 104 / 768
Heap size: 8634 / 100560 bytes

[00019973ms]
Thread: 0x20001E24, Stack size: 1784 / 4096
Thread: 0x20008D40, Stack size: 1272 / 2048
Thread: 0x20001F74, Stack size: 96 / 512
Thread: 0x20001F30, Stack size: 104 / 768
Thread: 0x200011EC, Stack size: 864 / 1200
Thread: 0x2000A000, Stack size: 768 / 816     <---
Heap size: 12598 / 100560 bytes

# Error

[00102271ms]
Thread: 0x20001E24, Stack size: 2216 / 4096
Thread: 0x20008D40, Stack size: 1272 / 2048
Thread: 0x20001F74, Stack size: 96 / 512
Thread: 0x20001F30, Stack size: 104 / 768
Thread: 0x200011EC, Stack size: 864 / 1200
Thread: 0x2000A000, Stack size: 768 / 816     <---
Heap size: 15690 / 100560 bytes

++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x8023C43
Error Value: 0x1
Current Thread: ppp Id: 0x2000A000 Entry: 0x8025FC5 StackSize: 0x330 StackMem: 0x2000A058 SP: 0x2001FF24
For more info, visit: https://mbed.com/s/error?error=0x80020125&tgt=WIO_LTE
-- MbedOS Error Info --

# .map

 .text._ZN4rtos6Thread6_thunkEPv
                0x08025fc4       0x40 BUILD/WIO_LTE/GCC_ARM/mbed-os/rtos/source/Thread.o
                0x08025fc4                rtos::Thread::_thunk(void*)

# Thread.cpp

void Thread::_thunk(void *thread_ptr)
{
    Thread *t = (Thread *)thread_ptr;

# mbed_config.h

#define MBED_CONF_PPP_THREAD_STACKSIZE                                        816                                                                                              // set by library:ppp

# ppp/mbed_lib.json

        "thread-stacksize": {
            "help": "Thread stack size for PPP",
            "value": 816
        },

```

matsujirushi, 2020-02-26:
I will try to test when increase stack size.

matsujirushi, 2020-03-06:
By setting the following, I got better.

mbed_app.json:
```
"ppp.thread-stacksize": 900
```

Thanks.

closed 2020-03-06 by matsujirushi