ARMmbed/mbed-os #12493: ThisThread::sleep_for() behaves differently since mbed od V 5.15.0

idea--list, 2020-02-22:
### Description of defect
The MAX32630FTHR board has an on-board RTC hardware. I change the register of this RTC in a way that it will tick only once per second (instead of 4096/second). This way the RTC will overflow only after 130+ years of usage.

Until mbed os 5.14.2 `ThisThread::sleep_for(1000);` results the code to wait 1 second just as expected.
However in mbed os 5.15.0 or 5.15.1 `ThisThread::sleep_for(1000);` results a really long interval and even `ThisThread::sleep_for(1);` results an 4 second long inactivity.

I really do not know what you guys are thinking when you change the behaviour completely leaving the name of the function unchanged and not informing us about that.

(I use Mbed for over a year now, but as you tend to keep change how many Mbed APIs work...Mbed is by far the most frustrating framework i have ever seen: code i write today might get broken tomorrow. Without any information about that.)

#### Target(s) affected by this defect ?
MAX32630FTHR

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio 0.8.1

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.15.0 or 5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?
Compile this code with mbed os 5.14.2 and 5.15.0 and run it on the target to reproduce the bug:

`#include "mbed.h"
#include "lib/max32630fthr/max32630fthr.h"
#include "lib/max32630fthr/MAX14690/MAX14690.h"
#include "pwrseq_regs.h"
#include "rtc_regs.h"

MAX32630FTHR pegasus;

Serial pc(USBTX, USBRX, 115200);

// main() runs in its own thread in the OS
int main()
{
     // let RTC run in all power modes
    MXC_PWRSEQ->reg0 |= MXC_F_PWRSEQ_REG0_PWR_RTCEN_RUN | MXC_F_PWRSEQ_REG0_PWR_RTCEN_SLP;

     // enable CPU to receive RTC interrupts
    MXC_CLKMAN->sys_clk_ctrl_1_sync = (1 << MXC_F_CLKMAN_SYS_CLK_CTRL_1_SYNC_SYNC_CLK_SCALE_POS);

    // tick RTC once per second
    MXC_RTCTMR->prescale = 12;

    // clear RTC
    MXC_RTCTMR->timer = 1582396775;

    // wait until changes get applied
    while((MXC_RTCTMR->ctrl & MXC_F_RTC_CTRL_PENDING) != 0);

    // enable RTC
    MXC_RTCTMR->ctrl |= MXC_F_RTC_CTRL_ENABLE;
    ThisThread::sleep_for(1);

    while (true) {
        int olvas = MXC_RTCTMR->timer;
        printf("rtc: %d\n", olvas);
        //ThisThread::sleep_for(1000);    /// OK in mbed 5.14.2
        //ThisThread::sleep_for(1);       // 4 sec in mbed 5.15.0
        wait_us(1000000);                 /// OK in mbed 5.15.0

        sleep();
    }
    return 0;
}`

idea--list, 2020-02-23:
The impact of this bug is really HUGE:
For example even the SDBlockDevice API malfunctions: if i reduce the frequency of the onboard RTC, the init() method of the SDBlockDevice API will not return anything before it comes to a timeout and the API will return error.

Not to mention any 3rd party code that do anything based on elapsed time.

0xc0170, 2020-02-24:
Firstly, there are couple of things that could have changed. As I read your reply referenced above (https://github.com/ARMmbed/mbed-os/pull/11463#issuecomment-590019432), this is a target issue. Good find! 
There was a fix that introduced issue to the target (not affecting other targets or generic layer in this case). I'll talk to the team to review this: @mbed-os-core @gpsimenos

> (I use Mbed for over a year now, but as you tend to keep change how many Mbed APIs work...Mbed is by far the most frustrating framework i have ever seen: code i write today might get broken tomorrow. Without any information about that.)

As this issue, if you have any other, please report. This is useful feedback but we need more concrete examples to do better the next time. In this case, it was simple fix that has consequences. We will investigate why tests for sleep did not see broken timings.

idea--list, 2020-02-25:
@0xc0170 

> This is useful feedback but we need more concrete examples to do better the next time.

I reported #11081 back in July. It took really long time to investigate what all you have changed in the USBDEVICE as of V5.10, mainly because you changed a bunch of things in each and every release without any documentation. On top of all that you also changed the location of the files without changing any code in the last round with USBDEVICE. I wrote a summary about my findings to Maxim Integrated, an engineer is working on it for months now and they found a bug recently... so the USBDEVICE is still broken for this target. Also that passed the testing process. Please contact and cooperate with Maxim Integrated as i think they got stuck.

In #11081 i also reported that this target has an on-board uSD-slot and since SD functionality became part of the mBed core the target.json should be updated accordingly as most people do not know how to handle all that via mbed.app.json.

Then i also [reported](https://github.com/ARMmbed/mbed-os-example-ble/issues/259) that none of the official mbed BLE examples work on this board. First i thought it was mbed core related bug, then i had to find out that the code of the example has been rewamped after December 2018 that has lead to a major regression. That also passed your tests... and there is hardly any activity in the BLE repo despite many other issues are reported there.

As i am not a professional programmer and your automated tests seem to pass many funny things i would not try to patch these for the general public, would be nice however to see the mBed and/or Maxim team to take up my findings and work on these issues. Thanks!

yagmurgok, 2020-02-26:
idea--list Hi,
We are currently working on USB stack issue. Mbed usb API has changed and need to port every board with USB to ew USB API. We are targerting ti finizalize within 2 weeks. For the BLE part we are converting from our BLE stack to ARM Cordio stack. it will take like 6 weeks to finalize

idea--list, 2020-02-26:
Hi yagmurgo,
Thanks for the update, i will stay tuned.

idea--list, 2020-03-13:
@yagmurgok 
Is there any progress on this?

idea--list, 2020-10-02:
It seems Mbed OS 2 is abandonned for long time now ([link](https://os.mbed.com/questions/87257/Is-mbed-OS-2-still-alive/))
The issue i reported here rendered the board almost useless above Mbed OS V5.14 as almost every codebase will contain `ThisThread::sleep_for()` and friends, causing completely unexpected timing issues on this target.

This mbed_app.json file seems to revert back to normal functionality above Mbed OS V5.14:
`{
"target_overrides": {
    "MAX32630FTHR": {
         "target.macros_remove": ["MBED_TICKLESS"],
         "target.features_add": ["STORAGE"],
         "target.components_add": ["SD"],
         "sd.SPI_MOSI": "P0_5",
         "sd.SPI_MISO": "P0_6",
         "sd.SPI_CLK": "P0_4",
         "sd.SPI_CS": "P0_7"
    }
}
}`

Actually it removes tickless mode for this target, which introduced the issue described in the title.
Further more it also addresses the issue that SD functionality became part of Mbed OS core around V5.10 but on this target one can not use that, as targets.json simply does not tell Mbed this board would have the capabilty.

Are there any converse effects if we disable tickless mode for this target until there is a working patch that would allow reenabling the feature?

idea--list, 2020-10-04:
Digged up the documentation, found [this](https://os.mbed.com/docs/mbed-os/v6.3/apis/power-management-sleep.html) and [this](https://os.mbed.com/docs/mbed-os/v6.3/porting/tickless-mode.html).
Again a great example of poor documentation. I mean if i am not intended to use `sleep()` in my code i will most probably not read its documentation. Thus i will never know, that Mbed OS will automatically trigger sleep() in case the core has nothing to do.

I also read the documentation of tickless mode, inside that i found adding `"overrides": {
    "tickless-from-us-ticker": true
}` to directly to targets.json might fix the problem i described in the first post even with tickless mode left enabled.
And indeed adding this solves the issue, while it is also clear, that this way my board will only use sleep instead of deep sleep.

But yet again the documentation is poor as even after reading the above mentioned links, i still do not know if deep sleep is only available when tickless mode is enabled based on the lpticker, or is deep sleep also available when tickless mode is disabled?
Or if tickless mode is disabled (so that scheduling will rely on SysTicks) only sleep can be used by me/automatic power managment?

If the documentation was good i would not have any questions after reading those 2 links.

IMHO you should combine all the info in that 2 links and put it also underneath the section called "Introduction to Mbed OS 6". That way you would provide a clear overview&understanding about Mbed OS runtime for anyone before they would begin reading about any APIs and begin coding in Mbed.

As i could resolve the original issue, i close this ticket.

closed 2020-10-04 by idea--list

idea--list, 2020-10-04:
Found the root cause of the problem:
I made my own code to initialize the on-board RTC and set the prescale value there.

Turns out that the file `/targets/TARGET_Maxim/rtc_api.c` already defines a function (void rtc_init(void)) for that purpose, so user code should just contain rtc_init() to initializing the the on-board RTC. As a last step rtc_init() will call init_rtc(), wich will automatically set the prescaler to the value defined on line 49. So to sum it up by changing line 49, compiling, uploading the bin file and then powercycling the board the issue can be avoided completely.

Maybe it would be a nicer solution if `rtc_init()` would accept the prescaler value as a parameter so that the user should not modify line 49 in rtc_api.c in order to get the RTC working as expected even if one changes the prescale value.